mavlink/MAVSDK-Python #360: Cannot takeoff my quadcopter with MAVSDK-python

PathumDananjaya, 2021-06-16:
I have built a quadcopter using pixhawk 2.4.8 flight controller and I am new to this drone industry. 

My transmitter is Flysky I6 and It's already connected to my flight controller. I have created a Ubuntu 20.04 (virtual box) inside the windows OS and installed python 3.8.5 on Ubuntu.

then I executed the following code,

    import asyncio
    from mavsdk import System

    SYSTEM_ADDRESS = "serial:///dev/ttyUSB0:57600"

    async def run():

        drone = System()
        await drone.connect(system_address=SYSTEM_ADDRESS)

        print("Waiting for drone to connect...")
        async for state in drone.core.connection_state():
            if state.is_connected:
                print(f"Drone discovered!")
                break

        print("-- Arming")
        await drone.action.arm()

        print("-- Taking off")
        await drone.action.takeoff()
     
        await asyncio.sleep(5)

        print("-- Landing")
        await drone.action.land()

    if __name__ == "__main__":
        loop = asyncio.get_event_loop()
        loop.run_until_complete(run())

when executing this code. I can see my quadcopter arming without any errors.

but it does not take off. it gives an error.

    Waiting for mavsdk_server to be ready...
    Connected to mavsdk_server!
    Waiting for drone to connect...
    Drone discovered!
    -- Arming
    -- Taking off
    Traceback (most recent call last):
      File "simple_arm.py", line 30, in <module>
        loop.run_until_complete(run())
      File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
        return future.result()
      File "simple_arm.py", line 21, in run
        await drone.action.takeoff()
      File "/home/chanaka/.local/lib/python3.8/site-packages/mavsdk/action.py", line 360, in takeoff
        raise ActionError(result, "takeoff()")
    mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: takeoff(); params: ()
    Exception ignored in: <function System.__del__ at 0x7f54bc13e5e0>
    Traceback (most recent call last):
      File "/home/chanaka/.local/lib/python3.8/site-packages/mavsdk/system.py", line 56, in __del__
      File "/home/chanaka/.local/lib/python3.8/site-packages/mavsdk/system.py", line 92, in _stop_mavsdk_server
    ImportError: sys.meta_path is None, Python is likely shutting down

Please help me. I appreciate your answers.

PathumDananjaya, 2021-06-16:
Ohh seems like there are few questions related to this problem in the closed section. I will try to solve the issue by looking into that. but if you need to give any advice, feel free to comment here. thank you

JonasVautherin, 2021-06-16:
Are you running PX4? 

> mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: takeoff(); params: ()

This says that the autopilot answered with "COMMAND_DENIED". Could mean that it is not ready to takeoff (e.g. no gps fix, or some preflight check failing), for instance.

PathumDananjaya, 2021-06-16:
Thanks for the reply and I am using Ardupilot firmware. I think that is the issue. if that is the case I have to install the PX4 firmware to my Pixhawk 2.4.8 flight controller through qground controller right? 🙂

JonasVautherin, 2021-06-16:
Yes, I think that installing PX4 should solve it :+1:

PathumDananjaya, 2021-06-16:
> Yes, I think that installing PX4 should solve it 👍

Thank you sir, I will try ☺️

c-subasinghe, 2021-07-24:
thank you and now everything going well. but I have a small problem that can I control a servo motor using MAVSDK?

julianoes, 2021-07-24:
@c-subasinghe pleases file a new ticket and provide some detail of what works, what does not, what output you see, what versions you're using etc.

closed 2021-07-24 by julianoes