mavlink/MAVSDK-Python #361: Timeout error

offjangir, 2021-06-16:
after executing one mission : 
mavsdk sever shows this:
[01:27:31|Debug] current: 0, total: 1 (mission_impl.cpp:824)
[01:27:36|Debug] MAVLink: critical: Operation timeout (system_impl.cpp:253)

and retrying to takeoff() or any other stuff giver the error : 
  File "multimav.py", line 122, in <module>
    loop.run_until_complete(D.Mis(mission_items0,mission_items1,mission_items2))
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "multimav.py", line 82, in Mis
    await drone.mission.upload_mission(mission_plan)
  File "/home/reaperj/.local/lib/python3.6/site-packages/mavsdk/mission.py", line 667, in upload_mission
    raise MissionError(result, "upload_mission()", mission_plan)
mavsdk.mission.MissionError: TIMEOUT: 'Timeout'; origin: upload_mission(); params: (<mavsdk.mission.MissionPlan object at 0x7f092a08d470>,)

can you explain what is wrong?

JonasVautherin, 2021-06-16:
Do you have some code to share that we could use to reproduce your error with SITL?

offjangir, 2021-06-16:
Yes, so  I was using mavsdk-python to make classes and  tryout swarm algos. I wanted to execture some missions so I made this code:
#!/usr/bin/env python3
import asyncio
from mavsdk import System
from mavsdk.mission import (MissionItem, MissionPlan)
import time
#general function to intiate the drone
class Drone:
	def __init__(self,ID):
		self.drone_ID = ID 
	#mission method
	async def Mis(self,mission_items0,mission_items1,mission_items2):
		for x in self.drone_ID :
			drone = System(mavsdk_server_address="localhost", port=(t+x))
			if x == 0 :
				mission_items = mission_items0
			elif x ==1 :
				mission_items = mission_items1
			elif x == 2 :
				mission_items = mission_items2
			else: pass

			await drone.connect()	
			print_mission_progress_task = asyncio.ensure_future(self.print_mission_progress(drone))
			running_tasks = [print_mission_progress_task]
			termination_task = asyncio.ensure_future(self.observe_is_in_air(drone, running_tasks))
			mission_plan = MissionPlan(mission_items)
			await drone.mission.set_return_to_launch_after_mission(True)
			await drone.mission.upload_mission(mission_plan)
			await drone.action.arm()
			await drone.mission.start_mission()
		await termination_task
	

	async def print_mission_progress(self,drone):
		async for mission_progress in drone.mission.mission_progress():
			print(f"Mission progress: "f"{mission_progress.current}/"f"{mission_progress.total}")

	async def observe_is_in_air(self,drone, running_tasks):

		was_in_air = False
		async for is_in_air in drone.telemetry.in_air():
			if is_in_air:
				was_in_air = is_in_air

			if was_in_air and not is_in_air:
				for task in running_tasks:
					task.cancel()
					try:
						await task
					except asyncio.CancelledError:
						pass
				await asyncio.get_event_loop().shutdown_asyncgens()
				return
	

if __name__ == "__main__":
	ID = [0,2] #ID
	t = 50040 #port no for first drone
	D = Drone(ID) #Class initialization
	mission_items0 = [] #Default mission item list for drone 0
	mission_items0.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	mission_items1 = [] #Default mission item list for drone 1
	mission_items1.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	mission_items2 = [] #Default mission item list for drone 2
	mission_items2.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	
	#-------------------------------Commands-------------------------------------#
	#takeoff
	loop = asyncio.get_event_loop()
	loop.run_until_complete(D.Mis(mission_items0,mission_items1,mission_items2))

offjangir, 2021-06-16:
#!/usr/bin/env python3
import asyncio
from mavsdk import System
from mavsdk.mission import (MissionItem, MissionPlan)
import time
#general function to intiate the drone
class Drone:
	def __init__(self,ID):
		self.drone_ID = ID 
	#mission method
	async def Mis(self,mission_items0,mission_items1,mission_items2):
		for x in self.drone_ID :
			drone = System(mavsdk_server_address="localhost", port=(t+x))
			if x == 0 :
				mission_items = mission_items0
			elif x ==1 :
				mission_items = mission_items1
			elif x == 2 :
				mission_items = mission_items2
			else: pass

			await drone.connect()	
			print_mission_progress_task = asyncio.ensure_future(self.print_mission_progress(drone))
			running_tasks = [print_mission_progress_task]
			termination_task = asyncio.ensure_future(self.observe_is_in_air(drone, running_tasks))
			mission_plan = MissionPlan(mission_items)
			await drone.mission.set_return_to_launch_after_mission(True)
			await drone.mission.upload_mission(mission_plan)
			await drone.action.arm()
			await drone.mission.start_mission()
		await termination_task
	
	#terminal mission printing
	async def print_mission_progress(self,drone):
		async for mission_progress in drone.mission.mission_progress():
			print(f"Mission progress: "f"{mission_progress.current}/"f"{mission_progress.total}")

	#terminal observation printing
	async def observe_is_in_air(self,drone, running_tasks):

		was_in_air = False
		async for is_in_air in drone.telemetry.in_air():
			if is_in_air:
				was_in_air = is_in_air

			if was_in_air and not is_in_air:
				for task in running_tasks:
					task.cancel()
					try:
						await task
					except asyncio.CancelledError:
						pass
				await asyncio.get_event_loop().shutdown_asyncgens()
				return
	

if __name__ == "__main__":
	ID = [0,2] #ID
	t = 50040 #port no for first drone
	D = Drone(ID) #Class initialization
	mission_items0 = [] #Default mission item list for drone 0
	mission_items0.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	mission_items1 = [] #Default mission item list for drone 1
	mission_items1.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	mission_items2 = [] #Default mission item list for drone 2
	mission_items2.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	
	#-------------------------------Commands-------------------------------------#

	loop = asyncio.get_event_loop()
	loop.run_until_complete(D.Mis(mission_items0,mission_items1,mission_items2))

offjangir, 2021-06-16:
'''
#!/usr/bin/env python3
import asyncio
from mavsdk import System
from mavsdk.mission import (MissionItem, MissionPlan)
import time
#general function to intiate the drone
class Drone:
	def __init__(self,ID):
		self.drone_ID = ID 
	#mission method
	async def Mis(self,mission_items0,mission_items1,mission_items2):
		for x in self.drone_ID :
			drone = System(mavsdk_server_address="localhost", port=(t+x))
			if x == 0 :
				mission_items = mission_items0
			elif x ==1 :
				mission_items = mission_items1
			elif x == 2 :
				mission_items = mission_items2
			else: pass

			await drone.connect()	
			print_mission_progress_task = asyncio.ensure_future(self.print_mission_progress(drone))
			running_tasks = [print_mission_progress_task]
			termination_task = asyncio.ensure_future(self.observe_is_in_air(drone, running_tasks))
			mission_plan = MissionPlan(mission_items)
			await drone.mission.set_return_to_launch_after_mission(True)
			await drone.mission.upload_mission(mission_plan)
			await drone.action.arm()
			await drone.mission.start_mission()
		await termination_task
	
	#terminal mission printing
	async def print_mission_progress(self,drone):
		async for mission_progress in drone.mission.mission_progress():
			print(f"Mission progress: "f"{mission_progress.current}/"f"{mission_progress.total}")

	#terminal observation printing
	async def observe_is_in_air(self,drone, running_tasks):

		was_in_air = False
		async for is_in_air in drone.telemetry.in_air():
			if is_in_air:
				was_in_air = is_in_air

			if was_in_air and not is_in_air:
				for task in running_tasks:
					task.cancel()
					try:
						await task
					except asyncio.CancelledError:
						pass
				await asyncio.get_event_loop().shutdown_asyncgens()
				return
	

if __name__ == "__main__":
	ID = [0,2] #ID
	t = 50040 #port no for first drone
	D = Drone(ID) #Class initialization
	mission_items0 = [] #Default mission item list for drone 0
	mission_items0.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	mission_items1 = [] #Default mission item list for drone 1
	mission_items1.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	mission_items2 = [] #Default mission item list for drone 2
	mission_items2.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	
	#-------------------------------Commands-------------------------------------#

	loop = asyncio.get_event_loop()
	loop.run_until_complete(D.Mis(mission_items0,mission_items1,mission_items2))

'''

offjangir, 2021-06-16:
#!/usr/bin/env python3
import asyncio
from mavsdk import System
from mavsdk.mission import (MissionItem, MissionPlan)
import time
#general function to intiate the drone
class Drone:
	def __init__(self,ID):
		self.drone_ID = ID 
	#mission method
	async def Mis(self,mission_items0,mission_items1,mission_items2):
		for x in self.drone_ID :
			drone = System(mavsdk_server_address="localhost", port=(t+x))
			if x == 0 :
				mission_items = mission_items0
			elif x ==1 :
				mission_items = mission_items1
			elif x == 2 :
				mission_items = mission_items2
			else: pass

			await drone.connect()	
			print_mission_progress_task = asyncio.ensure_future(self.print_mission_progress(drone))
			running_tasks = [print_mission_progress_task]
			termination_task = asyncio.ensure_future(self.observe_is_in_air(drone, running_tasks))
			mission_plan = MissionPlan(mission_items)
			await drone.mission.set_return_to_launch_after_mission(True)
			await drone.mission.upload_mission(mission_plan)
			await drone.action.arm()
			await drone.mission.start_mission()
		await termination_task
	
	#terminal mission printing
	async def print_mission_progress(self,drone):
		async for mission_progress in drone.mission.mission_progress():
			print(f"Mission progress: "f"{mission_progress.current}/"f"{mission_progress.total}")

	#terminal observation printing
	async def observe_is_in_air(self,drone, running_tasks):

		was_in_air = False
		async for is_in_air in drone.telemetry.in_air():
			if is_in_air:
				was_in_air = is_in_air

			if was_in_air and not is_in_air:
				for task in running_tasks:
					task.cancel()
					try:
						await task
					except asyncio.CancelledError:
						pass
				await asyncio.get_event_loop().shutdown_asyncgens()
				return
	

 if __name__ == "__main__":
    ID = [0,2] #ID
    t = 50040 #port no for first drone
    D = Drone(ID) #Class initialization
    mission_items0 = [] #Default mission item list for drone 0
	    mission_items0.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	mission_items1 = [] #Default mission item list for drone     mission_items1.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	mission_items2 = [] #Default mission item list for drone 2
    mission_items2.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	
	#-------------------------------Commands-------------------------------------#
    loop = asyncio.get_event_loop()
    loop.run_until_complete(D.Mis(mission_items0,mission_items1,mission_items2))

offjangir, 2021-06-16:
```
#!/usr/bin/env python3
import asyncio
from mavsdk import System
from mavsdk.mission import (MissionItem, MissionPlan)
import time
#general function to intiate the drone
class Drone:
	def __init__(self,ID):
		self.drone_ID = ID 
	#mission method
	async def Mis(self,mission_items0,mission_items1,mission_items2):
		for x in self.drone_ID :
			drone = System(mavsdk_server_address="localhost", port=(t+x))
			if x == 0 :
				mission_items = mission_items0
			elif x ==1 :
				mission_items = mission_items1
			elif x == 2 :
				mission_items = mission_items2
			else: pass

			await drone.connect()	
			print_mission_progress_task = asyncio.ensure_future(self.print_mission_progress(drone))
			running_tasks = [print_mission_progress_task]
			termination_task = asyncio.ensure_future(self.observe_is_in_air(drone, running_tasks))
			mission_plan = MissionPlan(mission_items)
			await drone.mission.set_return_to_launch_after_mission(True)
			await drone.mission.upload_mission(mission_plan)
			await drone.action.arm()
			await drone.mission.start_mission()
		await termination_task
	
	#terminal mission printing
	async def print_mission_progress(self,drone):
		async for mission_progress in drone.mission.mission_progress():
			print(f"Mission progress: "f"{mission_progress.current}/"f"{mission_progress.total}")

	#terminal observation printing
	async def observe_is_in_air(self,drone, running_tasks):

		was_in_air = False
		async for is_in_air in drone.telemetry.in_air():
			if is_in_air:
				was_in_air = is_in_air

			if was_in_air and not is_in_air:
				for task in running_tasks:
					task.cancel()
					try:
						await task
					except asyncio.CancelledError:
						pass
				await asyncio.get_event_loop().shutdown_asyncgens()
				return
	

if __name__ == "__main__":
	ID = [0,2] #ID
	t = 50040 #port no for first drone
	D = Drone(ID) #Class initialization
	mission_items0 = [] #Default mission item list for drone 0
	mission_items0.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	mission_items1 = [] #Default mission item list for drone 1
	mission_items1.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	mission_items2 = [] #Default mission item list for drone 2
	mission_items2.append(MissionItem(47.398039859999992,8.5455725400000000,25,10,True,float('nan'),float('nan'),MissionItem.CameraAction.NONE,float('nan'),float('nan')))
	
	#-------------------------------Commands-------------------------------------#

	loop = asyncio.get_event_loop()
	loop.run_until_complete(D.Mis(mission_items0,mission_items1,mission_items2))
```

offjangir, 2021-06-16:
running twice give you timeouts .. 
You can try the example code from the examples but there too after one successful mission same things happen
also, for connecting to gazebo. [#358 ](https://github.com/mavlink/MAVSDK-Python/issues/358)
was used as described by me

liuuuk311, 2021-09-19:
I'm follwing! I get the same error. 

I tried both with a `mission_raw.upload_mission()` by importing a mission plan from qGroundControl and by manually creating the MissionItems.

I'm using mavsdk  0.19.0

However, weirdly enough on the SITL it works, so I'm guessing that something is wrong with the firmware on my board (Matek F405-SE)

Can someone help me figure out what the issue is about?

JonasVautherin, 2021-09-20:
We have improved the mission upload recently in MAVSDK, let's see if it's better after we release a new version of mavsdk-python, I would say :+1:

liuuuk311, 2021-09-20:
Thank you @JonasVautherin that's good news! Just out of curiosity when the new version will be released?

JonasVautherin, 2021-09-30:
It's here! https://github.com/mavlink/MAVSDK-Python/releases/tag/0.20.0

closed 2026-10-06 by offjangir