mavlink/MAVSDK-Python #364: grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with: status = StatusCode.UNAVAILABLE details = "Socket closed"

MuhammadBilal1, 2021-07-02:
Hi,
I am using MAVSDK Python and getting the below error occasionally, I don't know why but this comes mostly at "Waiting for drone to connect"

mavsdk version: 0.16.0
grpcio version: 1.38.0
celery version: 5.1.0
Ubuntu 20.04 (Latest updates installed as of today)

**Note:** Mostly when I try to connect the drone, it kept waiting for the connecting at "Waiting for drone to connect..." then I use the below command to kill the mavsdk process and then connect again and it works. But this GRPC error comes even when I am not killing the ports manually.

sudo kill -9 $(sudo lsof -t -i:14540 -i:50051 )

[image]

JonasVautherin, 2021-07-02:
This happens when the connection between MAVSDK-Python and `mavsdk_server` is lost, I would say.

MuhammadBilal1, 2021-07-02:
Can you please tell me how we can fix this? And why this is happening?

[image]
[image]
[image]

JonasVautherin, 2021-07-02:
Did you manage to run the MAVSDK-Python example scripts?

MuhammadBilal1, 2021-07-02:
Yes, the examples work.
I am getting this issues occasionally in my flow, not all the times.

I am using the [Gazebo Headless](https://hub.docker.com/r/jonasvautherin/px4-gazebo-headless) simulator.

JonasVautherin, 2021-07-06:
How do you start `mavsdk_server`? Automatically with `await drone.connect()`?

MuhammadBilal1, 2021-07-08:
> How do you start `mavsdk_server`? Automatically with `await drone.connect()`?

Yes. Automatically with `await drone.connect()`

JonasVautherin, 2021-07-08:
Do you have some code you could share that would reproduce this issue?

MuhammadBilal1, 2021-10-01:
> Do you have some code you could share that would reproduce this issue?

@JonasVautherin  I can send you the code. Can't share it here with the public. My code has copyrights.
Please share the email or something where I can share this code.

JonasVautherin, 2021-10-01:
Could you isolate the bug into a small, reproducible example that you could share here?