mavlink/MAVSDK-Python #365: Connection fails if started while autopilot is on

amzntyshka, 2021-07-02:
I've built an application with mavsdk, and one thing I'm noticing is things work fine if the drone is turned on after the script starts, but the connection events do not work correctly if the drone is already on when the script is started. I made a simple asyncio subscriber to test this:

    async def recv_connection_state(self):
        async for state in self.drone.core.connection_state():
            if state.is_connected:
                print(f"Robot discovered!")
                self.connected = True
            else:
                print("Robot disconnected")
                self.connected = False

If I run this code with the robot already on, I get the following output:

Robot discovered!
Robot disconnected
Robot discovered!
Robot disconnected

If the robot is off when the script is started, I get a single

Robot discovered!

message. Not sure if it's relevant but I'm using a telemetry radio as my link. Is this known/intended behavior or a bug?

JonasVautherin, 2021-07-06:
Sounds like there is some work to do on the connection side, indeed

amzntyshka, 2021-08-04:
Additionally, I seem to be having issues with reconnection. Even if mavsdk is started first, the robot will properly connect on the initial power-up run, but if the robot is power cycled, the same connect-disconnected event triggers again. It's clear that the robot isn't actually disconnected, as it sends and receives MAVLink messages properly, it's just that there is a false disconnect event.

amzntyshka, 2021-08-04:
@JonasVautherin I have determined the issue is only present on 0.18 and above. Versions 0.17 and below have correct connect/disconnect behavior

julianoes, 2021-08-05:
I would assume disconnection here means that the robot disappears for a bit and "times out" but then comes back. Isn't that the case?

amzntyshka, 2021-08-05:
@julianoes The problem is disconnect and reconnect events are not one-to-one. There are more disconnect events than reconnect events. Either the disconnect events are false positives, or the robot is indeed reconnecting, but in that case there are missing connect events. So, for example, testing on 0.18.0, if I turn the robot off and on again, I get a disconnect when the robot turns off, a connect event immediately followed by a disconnect event when the robot turns on again, and then two disconnect events when the robot turns off again. This is 4 disconnect events and a single connect event. This means my program thinks the robot is disconnected when it is actually connected. If I test on 0.17.0, I get a disconnect when the robot turns off, a connect when the robot turns on, and a disconnect when the robot turns off again. This is one-to-one mapping and expected behavior

julianoes, 2021-08-05:
Oh ok, that sounds like a bug then! I will try to reproduce it but it might take some time. Maybe @JonasVautherin will get to it before me.

JonasVautherin, 2021-09-30:
This may be fixed in 0.20.0. @amzntyshka would you mind trying again with the latest MAVSDK-Python version?