mavlink/MAVSDK-Python #366: Never connect to:

alejandrovc19, 2021-07-15:
Hi to everyone! I have an Issue, when I want to run mavsdk on the NAVIO2, I have a problem it looks like this:
[image]

I already install the mavsdk-server manually and also run it manually before running mavsdk, with this command:

sudo apt-get install wget # if you don't already have wget
cd ~/Navio2/MAVSDK-Python/mavsdk/bin
wget https://github.com/mavlink/MAVSDK/releases/download/v0.40.0/mavsdk_server_linux-arm64-musl
 # change this for your specific architecture / release version
mv mavsdk_server_linux-arm64-musl mavsdk_server # Change the filename so it can be found
sudo chmod +x mavsdk_server # allow user to run, mine by default didn't have EXECUTE permissions

Then after running "apython"on MAVSDK-Python:

from mavsdk import System
drone = System(mavsdk_server_address="localhost")   "I put localhost because I install manually the mavsdk_server on the /Navio2/MAVSDK-Python/mavsdk/bin directory"
await drone.connect(system_address="udp://[192.168.1.76][:14500]")  "I have my Navio2 connect to that Ip, I already make a Mavlink connection to Mission Planner, but now I want to use MAVSDK but I don't now what is wrong.

JonasVautherin, 2021-07-16:
Multiple things here:

1. When you do `drone = System(mavsdk_server_address="localhost")`, it tells MAVSDK-Python to not start `mavsdk_server` automatically (so you have to do it manually, which you do). In that case you can just `await drone.connect()`, because the `system_address` is only used when MAVSDK-Python starts automatically.

In other words, your python code should look more like this:

```python
from mavsdk import System
drone = System(mavsdk_server_address="localhost")
await drone.connect()
```

2. When you run `mavsdk_server`, by default it will listen for heartbeats on UDP:14540. If your Navio2 broadcasts heartbeats on UDP:14500, you should run it like so:

```
./mavsdk_server_linux-arm64-musl udp://:14500
```

This is for the situation where the Navio2 initiates the connection (i.e. the Navio2 sends heartbeats to udp port 14500, and MAVSDK has to listen for them). If the Navio2 listens on 14500 for heartbeats coming from `mavsdk_server`, that would be:

```
./mavsdk_server_linux-arm64-musl udp://192.168.1.76:14500
```

But unfortunately that does not work yet with `mavsdk_server`. So for now you should make it such that the Navio2 broadcasts the heartbeats (as opposed to listening to heartbeats from MAVSDK). If you cannot do that, you can put `mavlink-router` in the middle to "invert" the connection initialization.

Does that make sense?

julianoes, 2021-07-24:
No follow up, closing.

closed 2021-07-24 by julianoes