mavlink/MAVSDK-Python #369: Change home position (RTL position)

schpatz, 2021-07-27:
I want to change the home position (RTL position) so that it is different from the starting position. I thought about just adding it as the last mission item and then execute the `action.land()` command but that would have some complications with other commands that still use the RTL position.
Next I looked at the TelemetryServer Class and tried this:
```
telserver = telemetry_server.TelemetryServer(async_plugin_manager.AsyncPluginManager())
await telserver.publish_home(home=telemetry_server.Position(lat_home, lon_home, 35,0))
```
But then i get this error (probably because I'm not initializing it correctly):
`AttributeError: 'AsyncPluginManager' object has no attribute '_channel'`

Would be really greatful for some insight or a better way to achieve this :)

julianoes, 2021-07-28:
Ok there are at least two ways how you can force PX4 to land somewhere else than home.

1. You set the home position to a new position using the command [MAV_CMD_DO_SET_HOME](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_HOME). This is currently not implemented in MAVSDK but would not be too hard. I'd think we could add it to the action plugin.
2. You use rally points which are explicitly made for the use case where you want "home positions" close to your drone. MAVSDK does not support rally points yet, however, the addition would not be too hard as the mechanism to add them is the same as adding mission items or geofence items. When using rally points, you can specify the behavior using the param [RTL_TYPE](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#RTL_TYPE).

Let me know which approach you would prefer and then I can try to guide you in how to add the functionality.

schpatz, 2021-07-29:
> 1. You set the home position to a new position using the command [MAV_CMD_DO_SET_HOME](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_HOME). This is currently not implemented in MAVSDK but would not be too hard. I'd think we could add it to the action plugin.

Thanks for the fast reply, that option sounds exactly like what I was looking for. I thought about that but didn't know how to send mavlink commands. I would be really greatful if you would walk me through how to implement the feature.

schpatz, 2021-07-31:
This seems really daunting and I will try to implement it, but it would be nice nevertheless if you or someone else tackles the problem as I am sure that some other people would benefit from this feature aswell :)

JonasVautherin, 2021-08-09:
Just a quick question: are we sure that MAV_CMD_DO_SET_HOME is implemented in PX4? Haven't checked, but I thought it was not

potaito, 2021-08-09:
@JonasVautherin just checked, MAV_CMD_DO_SET_HOME is really not implemented in PX4...

JonasVautherin, 2021-08-09:
Thanks a lot @potaito!

@schpatz: so the very first step is to get that implemented in PX4. Then we can talk about it in MAVSDK :+1:

julianoes, 2021-08-11:
Or we don't implement it and use rally points instead...

potaito, 2021-08-11:
The one use case for changing the home location I see is when using precision landing. Especially if one wants to land on a moving target and therefore update the initial search location for precision landing by changing the home position. 

But even if precision landing is not using rally points right now, we could also just modify precision landing to use them rather than implementing the home position change.

@schpatz Julian and Jonas are right. Try solving your problem with rally points. Let us know if there's still something missing.

kennethakiti1, 2022-03-21:
hi, I am quite new to mavlink and the entire Mavsdk environment, however I'm trying to implement the Mavsdk-python on a raspberry pi and i would like to know if there are any documentation or instructions sets i can follow to learn how to chang the home position of the px4

potaito, 2022-03-21:
@kennethakiti1 MAVSDK is currently not the limiting factor here, since PX4 does not have the implementation to change its home location: https://github.com/mavlink/MAVSDK-Python/issues/369#issuecomment-895280502

What behaviour do you want to achieve?

kennethakiti1, 2022-03-21:
@potaito I tried to run the takeoff_and_land.py example in the MAVSDK-Python [https://github.com/mavlink/MAVSDK-Python/blob/main/examples/takeoff_and_land.py](url), however I got the error 

`Drone discovered!
Waiting for drone to have a global position estimate...
Global position estimate ok
-- Arming
-- Taking off
Traceback (most recent call last):
  File "c:\Users\KAY\Downloads\MAVSDK-Python-main\MAVSDK-Python-main\examples\takeoff_and_land.py", line 37, in <module>
    loop.run_until_complete(run())
  File "C:\Users\KAY\AppData\Local\Programs\Python\Python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "c:\Users\KAY\Downloads\MAVSDK-Python-main\MAVSDK-Python-main\examples\takeoff_and_land.py", line 28, in run
    await drone.action.takeoff()
  File "C:\Users\KAY\AppData\Local\Programs\Python\Python39\lib\site-packages\mavsdk\action.py", line 360, in takeoff
    raise ActionError(result, "takeoff()")
mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: takeoff(); params: ()`

so from there, i decided to check the drone health using the code

```
#!/usr/bin/env python3`

import asyncio
from mavsdk import System

async def run():
    # Init the drone
    drone = System()
    await drone.connect(system_address="serial://COM9:115200")

    asyncio.ensure_future(print_health(drone))

async def print_health(drone):
    async for health in drone.telemetry.health():
        print(f" Drone Health: {health}")

if __name__ == "__main__":
    # Start the main function
    asyncio.ensure_future(run())

    # Runs the event loop until the program is canceled with e.g. CTRL-C
    asyncio.get_event_loop().run_forever()
```
and i got the output that all other conditions are okay except for the homeposition. so i thought setting the homeposition to that of the arming gps position world allow me to takeoff

`Drone Health: Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: True, is_global_position_ok: True, is_home_position_ok: False, is_armable: False]`

kennethakiti1, 2022-03-21:
@potaito please is it possible for me to initiate takeoff whiles is_home_position_ok is false and if so how can I go about it?

potaito, 2022-03-21:
Ah yes, I ran into the same issue. It seems that what MAVSDK is checking in the example apps to determine whether the drone is ready to takeoff or not, does not suffice. The error does not occur if you start the simulation, wait a bit and then start your python script. It's really only about timing, and not a user error. 

> and i got the output that all other conditions are okay except for the homeposition. so i thought setting the homeposition to that of the arming gps position world allow me to takeoff

I'm glad we are having this discussion, because setting the home position manually would be the completely wrong approach :sweat_smile: 
PX4 waits for a GPS lock first. This is the global_position check in MAVSDK. And only once GPS is acquired and has a sufficient quality, the PX4 home position is set automatically. The problem with the MAVSDK-Python examples is, that they are only checking the GPS lock of the drone. As soon as there is a lock, it tries to arm the drone. Often PX4 will not yet have set the home location, and will therefore deny arming. If the MAVSDK script waited just a second longer, the arm and takeoff would work just fine.

Here is a proposed fix: I modified one MAVSDK-Python example to check the global _and_ the home position, which turned out to be more reliable: https://github.com/mavlink/MAVSDK-Python/blob/108ef861f3af5fb414f3f84cb7132533cf288f1c/examples/follow_me_example.py#L31

It's my bad for not having it rolled out to in all the example scripts. I'll do that in a moment.

kennethakiti1, 2022-03-22:
@potaito, 

> if global_lock.is_global_position_ok and global_lock.is_home_position_ok: 

solves the issue of the error. However,  most of the time my drone isn't getting the home position, even though QGC indicates that the GPS can see 13 satellites.
[image]

and the output of the drone health and raw_gps using the code

```
#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():
    # Init the drone
    drone = System()
    await drone.connect(system_address="serial://COM9:115200")

    asyncio.ensure_future(print_health(drone))
    asyncio.ensure_future(print_raw_gps_info(drone))

async def print_health(drone):
    async for health in drone.telemetry.health():
        print(f" Drone Health: {health}")

async def print_raw_gps_info(drone):
    async for raw_gps in drone.telemetry.raw_gps():
        #print(f" RAw GPS info: {raw_gps}")
        gps_time = raw_gps.timestamp_us
        print(f"GPS Time :  {gps_time}")

        gps_alt = raw_gps.absolute_altitude_m # absolute altitude of drone in meters
        print(f"GPS ALT :  {gps_alt}")

        gps_lat = raw_gps.latitude_deg #latitude of the gps
        print(f"GPS LAT :  {gps_lat}")

        gps_lng = raw_gps.longitude_deg #longitude of the gps
        print(f"GPS LNT :  {gps_lng}")

if __name__ == "__main__":
    # Start the main function
    asyncio.ensure_future(run())

    # Runs the event loop until the program is canceled with e.g. CTRL-C
    asyncio.get_event_loop().run_forever()
```

outputs a result that indicates that home position is still false

```
GPS Time :  1664608185
GPS ALT :  51.78000259399414
GPS LAT :  30.1417383
GPS LNT :  119.9758004
 Drone Health: Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: True, is_global_position_ok: True, is_home_position_ok: False, is_armable: False]
```

so this makes me quite confused that after about 20 mins of having GPS directly under the sky, it still hasn't been able to get the home position(i must mention that it's currently raining but not heavily and the drone is only under an umbrella). 

are the 13 satellites not enough or is the weather the contributing factor here or am i still missing something

potaito, 2022-03-22:
13 satellites should definitely be enough. The drone should acquire a home position within a minute of powering on. I'm not sure if the umbrella is interfering though.

In any case, I don't think this is a MAVSDK issue, but rather PX4 in general.

kennethakiti1, 2022-03-23:
@potaito thanks, following your comment 

> In any case, I don't think this is a MAVSDK issue, but rather PX4 in general.

i decided to look more into the hardware and upon using a friend's GPS this problem seems to have been solved and now the home position is true

` Drone Health: Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: True, is_global_position_ok: True, is_home_position_ok: True, is_armable: False]`