mavlink/MAVSDK-Python #371: feature request: get current vtol state

mattes-bru, 2021-07-28:
I currently stumbled over the fact, that I can control the VTOL state from MAVSDK but don't have the possibility to read it's current state. Is this left out by purpose? I would have expected a property in the telemetry plugin, where the `in_air` property is generated from the extended_sys_state.

Regards

julianoes, 2021-07-29:
Good point. I think it would be worthwhile to add that to the telemetry plugin.

JonasVautherin, 2021-08-09:
Not sure I get what's missing. So when flying a VTOL, you don't know if it is `in_air` or not?

mattes-bru, 2021-08-10:
@JonasVautherin No, I get the inAir state, this is fine. I only wish I could get the current state of the vtol, wether it is currently flying as a fixed or rotary wing. 

@julianoes thank you for sharing this helpful links. I will have a deeper look after my vacation.

JonasVautherin, 2021-08-10:
Oh I see, so something like a `SubscribeVtolState` in `telemetry.proto` (from Julian's link above), right?

mattes-bru, 2021-08-10:
Exactly. This would be a really helpful extension. As far as I overlooked the API there is currently no way to determine the current vtol state.

JonasVautherin, 2021-08-10:
Yeah that makes sense :+1:. As Julian said, your first step is to add the function definition in `telemetry.proto` (I suggested `SubscribeVtolState`), and then generate the C++ code. Write the implementation there, and it will later come automagically to MAVSDK-Python :blush:. I made a live demo during the last DevSummit, if you want to have a look: https://www.youtube.com/watch?v=T1orxSyqDzI

mattes-bru, 2021-09-01:
@JonasVautherin @julianoes thank you both for your kind support. I currently made two pull requests mavlink/MAVSDK#1521 and mavlink/MAVSDK-Proto#250 Let me know when I need to add or change anything.

bazfp, 2021-09-02:
LGTM

closed 2021-09-02 by julianoes

mattes-bru, 2021-09-14:
@julianoes is there anything to do in order to have this feature available in mavsdk-python?

JonasVautherin, 2021-09-16:
I would suggest that we make a new mavsdk-python release after the next mavsdk release. Should automatically bring it then

mattes-bru, 2021-09-16:
Ok I found the issue, which stopped me from testing this: #391

julianoes, 2021-09-30:
FYI Release is coming https://github.com/mavlink/MAVSDK-Python/releases/tag/0.20.0