mavlink/MAVSDK-Python #375: Can not connect mavsdk server

Rikhsitilla, 2021-08-10:
Hi everyone, i had  a problem with mavsdk server

# !/usr/bin/env python3
import asyncio
from mavsdk import System
async def run():
    drone = System(mavsdk_server_address="127.0.0.1")
    await drone.connect(system_address="udp://127.0.0.1:14540")
here is my code 
I also used udp://:14540 ,udp://:14550,udp://:5600  
but did not work 
i have a following info 
    Waiting for mavsdk_server to be ready...
i could not connect my code to SITL .
Could you suggest any solution

JonasVautherin, 2021-08-10:
`udp://:14540` and `udp://:14550` are correct (maybe depending on your SITL config), but the other two are not.

Also using `mavsdk_server_address` means that you ran `mavsdk_server` manually. Did you? Otherwise what you want is:

```python
import asyncio
from mavsdk import System
async def run():
drone = System()
await drone.connect(system_address="udp://:14540")
```

The best to get started with MAVSDK-Python is probably to look at the examples, e.g. here: https://github.com/mavlink/MAVSDK-Python/blob/main/examples/takeoff_and_land.py#L9-L10

Rikhsitilla, 2021-08-10:
No i could not ran **mavsdk_server**manually . Because there is no such a file in mavsdk directory . Can you show how to ran automatically or manual

JonasVautherin, 2021-08-10:
> Can you show how to ran automatically or manual

What I showed above (and is in the examples linked above) runs mavsdk_server manually (if you are on a platform that supports it)

Rikhsitilla, 2021-08-11:
Actually when i am using example above i am getting following error:
Here is example code
'''#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():

    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered!")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break

    print("-- Arming")
    await drone.action.arm()

    print("-- Taking off")
    await drone.action.takeoff()

    await asyncio.sleep(5)

    print("-- Landing")
    await drone.action.land()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())'''

And here is a error.
**This installation does not provide an embedded 'mavsdk_server' binary.
If you installed using pip, this means that 'mavsdk_server' is not distributed
for your platform yet (e.g. arm).

You will need to get and run the 'mavsdk_server' binary manually:

  1. Download 'mavsdk_server' from: https://github.com/mavlink/mavsdk/releases
     or build it from source.

  2. Run it, e.g. on port 50051:
     './mavsdk_server -p 50051'

  3. Set the 'mavsdk_server_address' and port when creating the System:
     'drone = System(mavsdk_server_address='localhost', port=50051)'**

I am using Ubuntu 18.04. I build **mavsdk_server**  from source , but can not run **./mavsdk_server -p 50051'**, I think problem is releted ro directory . Where i should  run it ?

closed 2021-08-11 by Rikhsitilla

JonasVautherin, 2021-08-12:
Did you solve your issue?

Note that instead of compiling from source, you can just get the binary here: https://github.com/mavlink/MAVSDK/releases/download/v0.42.0/mavsdk_server_musl_x86_64