mavlink/MAVSDK-Python #377: USB connection windows10 Imu information

JuanshuB, 2021-08-10:
Platform: Windows 10 and CubeOrange 
Connection: USB (COM 9 (SLCAN) and COM10 (Mavlink))

I am trying to get the IMU information of my CubeOrange in real time in my laptop to syncronize that data with other sensors, so I think MAVSDK - python is what I am looking for. Firstly I downloaded Ardupilot and did the connection so I can see the values of the imu sensors in Ardupilot via COM10 and 57600. Nevertheless, I dont know how to do the configuration to connect the CubeOrange with Python in the script "takeoff_and_land" :

#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():

    drone = System()
    await drone.connect(system_address="serial:///dev/ttyCOM10:57600")
    
    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered!")
            break

Should I configure "drone = System()"? What serial port works with Windows?

JonasVautherin, 2021-08-10:
First, note that support is limited with Ardupilot, but I think it should connect successfully.

Then I don't use Windows, but it seems to me that `/dev/ttyCOM10` is not exactly a path for Windows. Or is it? How do you connect from QGC? And are you able to create a "Comm Link" in QGC for that?

JuanshuB, 2021-08-11:
Hi. I don't think "/dev/ttyCOM10" is a path for Windows, but I don't know what could be (I have also tried with "/dev/ttyUSBx", "/dev/ttySx" and "/dev/ttyACMx" with x from 0 to 10, but unsuccessfully). And secondly, I have not connected it with QGC so nor creating a "Comm Link". What I have done is to install Ardupilot in my laptop via this link (https://docs.cubepilot.org/user-guides/autopilot/the-cube-series-firmware-and-connection-troubleshooting) and installing in the CubeOrange the firmware "Copter 4.0.7 OFFICIAL"... Am I skipping important steps?

JonasVautherin, 2021-08-12:
I would first try to connect with QGC, and then to find the correct path to use on Windows :+1:

JuanshuB, 2021-08-13:
Hi Jonas, sorry for the delay. First of all, I have tried the connection in my Nvidia Jetson Nano, and the drone was discovered, but the CubeOrange is not able to send messages with Python, and I discovered it is because I have Ardupilot and not PX4. So I am trying now to install PX4 in my CubeOrange using QGC in W10, but it does not work. I install the drivers which come with QGC, but they don't seem to be suitable because the COM port of the MAVLINK is not detected, and I can not connect QGC and my CubeOrange working PX4. 

How have I install PX4 so? If a uninstall the driver, W10 recognize the USB like a normal, so QGC let me install PX4, then I install the drivers, and the same problem as before. In the Jetson nano, it keeps waiting a connection with the USB, (it doesnot give an error but neither a drone discovery).

 I suppose that if I am able to connect QGC and CubeOrange in W10, it would also work in the Nano, which is what a really need.
Thanks in advance

julianoes, 2021-08-30:
@JuanshuB I'm sorry but @JonasVautherin or I probably can't help with Windows and Cube problems, as we don't have Cubes or Windows :smile:.

However, feel free to ask in the PX4 Slack.

closed 2021-08-30 by julianoes

MaEtUgR, 2025-01-06:
For the record if someone is reading this in the future. I can successfully connect on Windows using Python with:
```
drone = System(mavsdk_server_address='localhost', port=50051)
await drone.connect(system_address="serial://COM10")
```
although I think what counts is to prepare by running the latest "win32" build of mavsdk_server downloaded from here: https://github.com/mavlink/mavsdk/releases using the command:
```
mavsdk_server_win32.exe serial://COM10
```

Note: I'm testing with an original Pixracer connected over USB.