mavlink/MAVSDK-Python #378: takeoff_and_land.py Error: mavsdk.action.ActionError: CONNECTION_ERROR: 'Connection Error'; origin: arm();: ()

alejandrovc19, 2021-08-10:
Hi to everyone!
I want to run the example takeoff_and_land.py and that problem happened me, and I don't now how to fix that, I've already run the telemetry.py examples and everything is working fine, I'm running the PX4v2 on a Raspberry pi 4 B, with Telemetry connection(TELEM2) to USB on the raspberry, port ttyACM0 

[image]

Also, I make a UDP connection to my computer the GS with mission planner and it is working fine

[image]

But the problem became when I want to run the examples, someone can help me?

alejandrovc19, 2021-08-10:
I'm running this example: 

[image]

mrpollo, 2021-08-12:
@alejandrovc19 as discussed MAVSDK doesn't yet support ArduCopter, if you run into any issues with PX4 let us know.