mavlink/MAVSDK-Python #383: Feature Request: MAV_CMD_DO_SET_RELAY/SERVO (181-184)

DoppleGangster, 2021-08-18:
I was going to add these plugins with a pull request. Wanted to confirm if adding them to the Action plugin would be the correct location.

[Set Relay](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_RELAY)
[Set Relay Repeat](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_REPEAT_RELAY)
[Set Servo](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_SERVO)
[Set Servo Repeat](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_REPEAT_SERVO)

Thanks!

JonasVautherin, 2021-08-23:
Are those implemented in PX4? Can you explain what a "relay" is? And how are those different from the already-supported [MAV_CMD_DO_SET_ACTUATOR](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ACTUATOR) (see [here](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/action/action.proto#L101-L104))?

DoppleGangster, 2021-08-23:
[This ](https://docs.px4.io/master/en/payloads/#mission-triggering)section of the documentation is what led me to make the request. 

I'll use the command from the link you sent. Seems the set relay command would be somewhat redundant.

Thanks for you help.

DoppleGangster, 2021-09-23:
Perhaps I'm not interpreting the mixer file correctly, but is there no way to have an output act as a digital output? 3.3V when set high and 0V when low? Does the mavlink command for set relay provide this?