mavlink/MAVSDK-Python #384: Real Pixhawk how to get system_address

hanlu1215, 2021-08-19:
In the official examples, the simulation address is used, but if I want to connect to the real flight control, how do I need to get the address, what should be written。
[image]

DoppleGangster, 2021-08-20:
How are you connecting to the flight board?

julianoes, 2021-08-30:
I'm closing this. Please comment again if it is still not clear.

closed 2021-08-30 by julianoes

FarziBuilder, 2024-02-23:
Obviously it is still not clear. Please tell someone how to get the system_address?

JonasVautherin, 2024-02-23:
The `system_address` is the interface on which you want to listen to MAVLink messages coming from the drone. You have to know how you expect the MAVLink messages to move between your drone and MAVSDK in your setup (is it UDP, is it serial, is it a carrier pigeon?).

Please open a new question and detail your setup if you want us to help, this one was a question from what would probably a different setup, years ago.