mavlink/MAVSDK-Python #385: CAN'T ARM THE DRONE

BrenoPimentel, 2021-08-21:
Hi, let me try to explain my issue.
I use AirSim for my drone's simulations. To use AirSim, i have installed PX4 simulator and run it on WSL 2. Everything works fine with these two softwares. I also included mavSDK to communicate with PX4 to fly the drone on AirSim, however, when I use the command line "drone.action.arm()", the following error appears to me:
Traceback (most recent call last):
  ### File "C:\cursopython\teste_colisao.py", line 26, in <module>
    loop.run_until_complete(print_flight_mode())
  File "C:\Users\breno\AppData\Local\Programs\Python\Python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "C:\cursopython\teste_colisao.py", line 15, in print_flight_mode
    await drone.action.arm()
  File "C:\cursopython\venv\lib\site-packages\mavsdk\action.py", line 312, in arm
    raise ActionError(result, "arm()")
mavsdk.action.ActionError: CONNECTION_ERROR: 'Connection Error'; origin: arm(); params: ()
Exception ignored in: <function System.__del__ at 0x000001D695E77EE0>
Traceback (most recent call last):
  File "C:\cursopython\venv\lib\site-packages\mavsdk\system.py", line 56, in __del__
  File "C:\cursopython\venv\lib\site-packages\mavsdk\system.py", line 92, in _stop_mavsdk_server
ImportError: sys.meta_path is None, Python is likely shutting down

Note: I try to run mavSDK and AirSim on windows, and PX4 on WSL 2.

the code is:
`import asyncio
from mavsdk import System

async def print_flight_mode():
    drone = System()
    await drone.connect(system_address="udp://:14580")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered!")
            break
    
    await drone.action.arm()

    await drone.action.takeoff()
    await drone.action.hold()

    async for flight_mode in drone.telemetry.flight_mode():
        print("FlightMode:", flight_mode)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(print_flight_mode())`

JonasVautherin, 2021-08-23:
It's failing because `arm()` is called when the drone is disconnected, apparently. Does it say "Drone discovered!" before that?

BrenoPimentel, 2021-08-23:
Yes!! 
The full output is: 
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
Drone discovered!
Waiting for drone to have a global position estimate. . .

After "waiting for drone to have a global position estimate. . .", nothing happens
If i delete the commands:
`print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered!")
            break` 
or use `await drone.action.arm()` before it,  the error that i have described before, appears to me.

BrenoPimentel, 2021-08-23:
Just to make myself clear. I have set up TCP port 4560 and UDP port 14540. 
On AirSim setting, set up:
```
"ControlIp": "172.31.66.156",
"ControlPortLocal": 14540,
"ControlPortRemote": 14580,
"LocalHostIp": "172.31.64.1",
```
ControlIp: I have used my scope global eth0 WSL 2 IP
LocalHostIP: my Ethernet vEthernet (WSL) IP

JonasVautherin, 2021-08-23:
Did you try with SITL-gazebo, to get used to it before trying with Airsim? [This](https://github.com/jonasvautherin/px4-gazebo-headless) may be a quick way to try

JonasVautherin, 2021-08-24:
Does it work with QGC?

BrenoPimentel, 2021-08-24:
Yes it works!!
Thanks Jonas
I work on @Black Bee Drones team, after trying hard we figured out how to solve it.
Seems that PX4 just send Malink signal to localhost, so we were just able to connect it by running mavSDK on WSL 2 instead of windows. On that matter, i had to search, over PX4 settings ```mavlink_main.h```, and figure out that its possible to enable mavlink broadcast to all network. Therefore, we had to enable the broadcast, by changing the commnad line ```BROADCAST_MODE_OFF``` to. ```BROADCAST_MODE_ON```. 
At first sight, everything works fine, however, we noticed that mavSDK can only connect UDP port 14550. It may be a problem when running multiples drones simulation, not sure. I am searching to understand if its a mavSDK problem or PX4

JonasVautherin, 2021-08-24:
Ok, good to read that this is fixed :+1:

closed 2021-08-24 by JonasVautherin