mavlink/MAVSDK-Python #386: CAN'T CONNECT WITH AIRSIM

RyanFoward, 2021-08-23:
Hello Jonas! I'm Ryan, and I'm part of a university team that works with autonomous drones.
We've had problems with connect mavsdk on windows to the simulator AirSim, we are using this lines of code:
```
drone = System()
    await drone.connect(system_address="udp://:14540")
```
but the only message that appears is:
```
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
```
but never connect with the drone

What can we try?

closed 2021-08-23 by RyanFoward

JonasVautherin, 2021-08-24:
Did you find a solution?

RyanFoward, 2021-08-24:
Yes!! We find a way to do this connection! Thx Jonas!

JonasVautherin, 2021-08-24:
Would you mind explaining here what was wrong and how you fixed it? So that it could help somebody else with the same issue? That's how it usually works :innocent:: you ask for help, and if you find the answer before somebody else can help you, you share it with the community :+1: