mavlink/MAVSDK-Python #387: How to get messages/warning/errors in MAVSDK?

MuhammadBilal1, 2021-08-30:
Hi,
Is there a way in MAVSDK that can help us in getting the message/warnings/errors about the drone as we get in the QGroundControl/Mission planner?

For Example:
Critical Navigation Failure
Arming Denied
Failsafe enabled
etc.

Some messages can be seen on the attached images of QGC.

[image]

[image]

[image]

julianoes, 2021-08-30:
Try subscribing to status texts: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html?highlight=statustext#mavsdk.telemetry.Telemetry.status_text

MuhammadBilal1, 2021-08-30:
> Try subscribing to status texts: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html?highlight=statustext#mavsdk.telemetry.Telemetry.status_text

This is what I was looking for. **It worked.**

Thank you.

closed 2021-08-30 by MuhammadBilal1