mavlink/MAVSDK-Python #396: MAVSDK not compatible with Python version 3.10.0

Pascal91, 2021-10-17:
The following code shall be corrected in "`__init__.py`" to be compatible with **3.10.0**:
```
# Check for compatibility
if float(".".join(platform.python_version_tuple()[0:-1])) < 3.6:
    print("[!] MAVSDK-Python is only available on Python >= 3.6")
    sys.exit(1)
```

_NB: This is a usual coding/checking error with the **X.Y.Z** notation._

julianoes, 2021-10-18:
@Pascal91 would you mind creating a pull request with the change?

Pascal91, 2021-10-18:
I'm sorry, I'm not skilled enough with gitHub to pull a request ; here is the correction:
`if (sys.version_info.major >= 3) and (sys.version_info.minor >= 6):`

_NB: I would not call this issue an "enhancement" but a "bug" ; since it's really an implementation bug as as long as this is not corrected, MAVSDK will not work with Python 3.10._

julianoes, 2021-10-18:
@Pascal91 https://docs.github.com/en/github/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

But if you don't have time right now, no problem, we'll get to it.

However, could you elaborate what changes in 3.10 that we're not compatible with? We probably would like to actually fix that.

Pascal91, 2021-10-18:
Read again my previous posts here above ; it's just your "`Check for compatibility`" test at the beginning of "`__init__.py`" which is not correctly implemented.
You should correct the "`if`" with something like my proposal.