mavlink/MAVSDK-Python #402: Mission Upload not working with MAVSDK-Python on Ardupilot

iamvamsii, 2021-11-03:
I had issue in connecting to the vehicle's Pixhawk using target system ID with dronekit and pymavlink as there are other mavlink systems connected to it. But it seems both dronekit and pymavlink doesn't supports mentioning target system ID to the vehicle connect object.

However connecting with mavsdk-python solved the issue and I am able to only listen to the Pixhawk

But.. as far as I tested with mavsdk-python examples, only the arm, land and some telemetry read is working

Unable to takeoff and upload mission to the Ardupilot using MAVSDK-PYTHON... Did anyone tried uploading missions to Ardupilot?

JonasVautherin, 2021-11-03:
@bazfp made it work, but with the `mission_raw` plugin, not the `mission` plugin. So I'm afraid it's only available in C++ :confused:

julianoes, 2022-04-05:
No follow up, closing. And yes ArduPilot support is not complete, check https://github.com/mavlink/MAVSDK/issues/728 for reference.

closed 2022-04-05 by julianoes