mavlink/MAVSDK-Python #404: Unable to subscribe to ATTITUDE_TARGET(#83) updates

danielfhnw, 2021-11-05:
I'm currently trying to implement an autotune of the attitude controller of the PX4 (following picture).
To be able to tell how good the controller performs, **I need to get the setpoint (red circle)** and the corresponding actual value (green circle).

[image]

With the following code I'm able to connect to the PX4 and get the attitude angular velocity vector through MAVSDK-Python but not the setpoints.

```
import asyncio
from mavsdk import System

async def print_omega(drone):
    async for omega in drone.telemetry.attitude_angular_velocity_body():
        print(omega)

async def run():
    drone = System()
    await drone.connect(system_address="serial://COM4:115200")
    asyncio.ensure_future(print_omega(drone))

def test():
    asyncio.ensure_future(run())
    asyncio.get_event_loop().run_forever()
```

I know that the message is there as I can see the correct values in my QGroundControl (red box).

[image]

I have also searched the MAVLINK Docs and found the following standard.

[image]

Have I overlooked something in the telemetry.py file?
Is it not implementet? (I couldn't find a subscription to the setpoint vector in the C++ library either)
Does anybody know another way of getting the setpoint values?

Any Help is greatly appreciated.

julianoes, 2021-11-05:
I think that's just not implemented in MAVSDK. Feel free to make a PR to add it. First would be proto, then the C++ implementation, and then a new Python release.
See [docs](https://mavsdk.mavlink.io/main/en/cpp/contributing/plugins.html#add-api-to-proto) on how to do that.

Talking of auto-tune, have you seen? https://github.com/PX4/PX4-Autopilot/pull/18311

danielfhnw, 2021-11-12:
Thank you for your fast response.
Maybe I can find another way to easily get the values and if not I will come back and make a PR. 

As this autopilot was not fully integrated at the start of my project work, I need to do it myself. But I will surely dig into it to get some inspiration.

danielfhnw, 2021-12-09:
Using pymavlink I was able to get the desired ATTITUDE_TARGET Message with the following code:

```
from pymavlink import mavutil

def get_attitude_target(m):
    msg = m.recv_match(type='ATTITUDE_TARGET', blocking=True)
    print(msg)

master = mavutil.mavlink_connection("COM4")

while True:
    get_attitude_target(master)

```

Following is one of the received messages, to confirm that it is the correct one.

> ATTITUDE_TARGET {time_boot_ms : 687531, type_mask : 0, q : [0.8870108127593994, -0.0, 0.4617486000061035, 0.0], body_roll_rate : -0.15904225409030914, body_pitch_rate : 1.0471975803375244, body_yaw_rate : -0.14586926996707916, thrust : 0.5546696782112122}

It might not be the most beautiful solution, but it works for now.

closed 2021-12-09 by danielfhnw