mavlink/MAVSDK-Python #406: Serial Connection to Jetson Xavier

dimikout3, 2021-11-23:
I am trying to connect via serial (ttyUSB0) from pixhawk to a jetson xavier. I have installed mavsdk-python via pip3 and I am trying to execute examples/takeoff_and_land.py. With the only difference that 

`await drone.connect(system_address="udp://:14545")
`

has been changed to 

`await drone.connect(system_address="serial:///dev/ttyUSB0")
`
However I am getting the following error
```
This installation does not provide an embedded 'mavsdk_server' binary.
If you installed using pip, this means that 'mavsdk_server' is not distributed
for your platform yet (e.g. arm).

You will need to get and run the 'mavsdk_server' binary manually:

  1. Download 'mavsdk_server' from: https://github.com/mavlink/mavsdk/releases
     or build it from source.

  2. Run it, e.g. on port 50051:
     './mavsdk_server -p 50051'

  3. Set the 'mavsdk_server_address' and port when creating the System:
     'drone = System(mavsdk_server_address='localhost', port=50051)'
```

Any ideas why this is happening ? 
P.S I have checked that the FTDI is connected to ttyUSB0. I have enabled the mavlink connection (MAV_1_CONFIG = TELEM 2). I am using a HITL approach via AirSim (Although I believe that this is not relevant)

dimikout3, 2021-11-23:
Had some progress it seems that the Jetson Xavier has aarch64, that means that mavsdk_server should be initiated mannualy and connecte it to the serial port where the FTDI is. So I run 
`sudo ./mavsdk_server_linux-arm64-musl -p 50051 serial:///dev/ttyUSB0
`
and the result is something like 

```
[05:23:07|Info ] MAVSDK version: v0.48.0 (mavsdk_impl.cpp:28)
[05:23:07|Info ] Waiting to discover system on serial:///dev/ttyUSB0... (connection_initiator.h:20)

```

Then I run the takeoff_and_land.py  by connecting it to the server 

changed the line 
```
drone = System()
await drone.connect(system_address="udp://:14545")
```
to
```
drone = System(mavsdk_server_address='localhost', port=50051)
await drone.connect()
```
and got 

`Waiting for mavsdk_server to be ready...`

dimikout3, 2021-11-23:
Similar to #189

dimikout3, 2021-11-23:
Tried to test if I can connect to the pixhawk via mavlink by running the mavlink_shell.py from the PX4/tools repository.

`python3 mavlink_shell.py /dev/ttyUSB0 --baudrate 11520` 

and the script failed with exit error. 

`serial.serialutil.SerialExceptionL: device report readiness to read but returned no data (device disconnected or multiple access on port?)`

Where should I look to isolate the issue ?

JonasVautherin, 2021-11-23:
I think that if `./mavsdk_server_linux-arm64-musl -p 50051 serial:///dev/ttyUSB0` does not detect a drone, you have a problem there already. So the first thing would be to understand why mavsdk_server does not receive any mavlink messages when connecting to `/dev/ttyUSB0` :+1:

dimikout3, 2021-11-23:
Thanks for the fast response and the awsome work maintaining the repo! And @JonasVautherin you are right, it seems that despite I have the set MAV_1_CONFIG = TELEM 2 the Telemetry 2 port did not work .... I connected the FTDI to TELEMETRY 1 and everything worked correctly ! Now the question is to understand how can I activate both telemetry ports (I need one for the Jetson and one for radio telemetry data (via Q Ground Control station) ). I will keep you posted regarding my progress and please share any thoughts you may have....

cthaun14, 2022-01-21:
Have you tried to specify your baud rate for the serial connection when starting mavsdk-server. That is usually required.

cthaun14, 2022-01-21:
Red indicates that the parameter has been changed from Default. Have you been able to solve running both telem ports?

dimikout3, 2022-01-21:
Unfortunately  I did not solve it yet. But I will check it again as soon as possible.

julianoes, 2022-04-05:
Couple of things to try:
1. Follow these steps: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues
2. Check you're looking at the correct ttyUSBx device. Does it disappear when you disconnect the cable? And appear again when you plug it in?
3. Can you see garbled bytes arriving on it? I usually test that using `screen /dev/ttyUSB0 BAUDRATE.

Also, I just checked above and I wonder what baudrate you're using.

> python3 mavlink_shell.py /dev/ttyUSB0 --baudrate 11520

This doesn't look like a correct baudrate. A valid one would be 115200.

dimikout3, 2022-08-04:
@keytheengineer I managed to activate both TELEM1 (companion computer) and TELEM2 (radio telemetry to Qground control station) using the following parameters:

MAV_0_CONFIG TELEM1
MAV_0_FORWARD 1
MAV_0_MODE Normal
MAV_0_RATE 1200 B/s
SER_TEL1_BAUD 57600 8N1

MAV_2_CONFIG TELEM2
MAV_2_FORWARD 1
MAV_2_MODE Normal
MAV_2_RATE 1200 B/s
SER_TEL2_BAUD 57600 8N1

The radio telemetry works fine. However, the connectivity with the companion computers is far from perfect. It seems that the messages take many seconds to be sent. Overall, the mavsdk-server does not perform efficiently when there is another mavlink instance activated @julianoes any ideas on that issue?

julianoes, 2022-08-04:
> However, the connectivity with the companion computers is far from perfect.

Ok, first thing is that I would use a higher baudrate as that will give you more bandwidth for all the messages. E.g. 230400 or 921600.

> It seems that the messages take many seconds to be sent.

Post the output of mavsdk_server please. And make sure you use the latest mavsdk_server 1.4.4.

> Overall, the mavsdk-server does not perform efficiently when there is another mavlink instance activated

How is that all connected? Please give more context.

julianoes, 2023-02-10:
This could be fixed with https://github.com/mavlink/MAVSDK-Python/releases/tag/1.4.4..