mavlink/MAVSDK-Python #410: Connect multiple clients to one Mavsdk server

35129239SchellingerJim, 2021-12-03:
Good evening, I am failing with connecting multiple mavsdk clients (python) to the same mavsdk server which is connected to a single drone. 

E.g if I run the following code, but "client one" misses any telemetry as soon as "client two" connects: 

```
async def main_tel(*args):
    drone = System()
    print(f"Client {args[0]} wait for drone")
    await drone.connect(system_address="udp://:14540",)
    print(f"Client {args[0]} connected")
    async for mode in drone.telemetry.flight_mode():
        print(f"Client {args[0]} got mode {mode}")

def start_client(*args):
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main_tel(*args))

if __name__ == "__main__":
    p = Process(target=start_client,args=["client one"])
    p.start()
    time.sleep(3)
    print("Start Client two in new process")
    p = Process(target=start_client, args=["client two"])
    p.start()
    p.join()

```
produces the output: 

client one wait for drone
client one connected
client one got mode RETURN_TO_LAUNCH
client one got mode RETURN_TO_LAUNCH
Start Client two in new process
client two wait for drone
client two connected
client two got mode RETURN_TO_LAUNCH
client two got mode RETURN_TO_LAUNCH
client two got mode RETURN_TO_LAUNCH
...

Is this behavior may related to ?: 
https://github.com/mavlink/MAVSDK/issues/1637

Thanks in advance!

julianoes, 2021-12-04:
One mavsdk_server supports connecting to only one vehicle. For multiple vehicle you have to either use multiple mavsdk_server instances, or use C++.

That being said there is work going on for direct Python wrappers without the gRPC mavsdk_server but Python directly binding to the C++ API. This would then enable connecting to multiple vehicles.

35129239SchellingerJim, 2021-12-05:
Thank you for your reply, 
My Issue is not about connecting to multiple drones but one mavsdk server connected to one drone and two grpc clients (stubs) connecting to that mavsdk server from two different processes. 

I guess one grpc server should be abel to interact with two stubs but is this implemented in the python client?

JonasVautherin, 2021-12-06:
That will be a problem because right now the implementation only supports one subscription at a time. So for instance if the first stub subscribes to position updates, and then the second stub subscribes as well, then the first one will be unsubscribed.

There is an effort going on to improve that on the C++ implementation side, but that's not there yet (see [here](https://github.com/mavlink/MAVSDK/issues/1637))

closed 2022-04-05 by julianoes

ShafSpecs, 2024-03-04:
> That being said there is work going on for direct Python wrappers without the gRPC mavsdk_server but Python directly binding to the C++ API. This would then enable connecting to multiple vehicles.

Hi @julianoes :wave:. Is there any update on this? Using multiple vehicles with one connection and differentiating them via their ids