mavlink/MAVSDK-Python #411: error encountered running mission.py - mavsdk-python

binduramarao, 2021-12-03:
Hello:

I am just trying to test MAVSDK-Python. When I just run mission.py (simple jmdvsim and SITL environment), I get the errors:
All I did was:  **python3 mission.py**

Drone discovered!
Traceback (most recent call last):
  File "mission.py", line 111, in <module>
    loop.run_until_complete(run())
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "mission.py", line 27, in run
    mission_items.append(MissionItem(47.398039859999997,
TypeError: __init__() missing 1 required positional argument: 'camera_photo_distance_m'
^CException ignored in: <module 'threading' from '/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/threading.py'>
Traceback (most recent call last):
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/threading.py", line 1388, in _shutdown
    lock.acquire()

Any help is appreciated.

Bindu
busybrr@yahoo.com

CY-1992, 2021-12-04:
MissionItem requires you to pass one more additional argument as mission item API has been updated recently. 
try the below code

```python
    mission_items.append(MissionItem(47.398039859999997,
                                     8.5455725400000002,
                                     25,
                                     10,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan')))
    mission_items.append(MissionItem(47.398036222362471,
                                     8.5450146439425509,
                                     25,
                                     10,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan')))
    mission_items.append(MissionItem(47.397825620791885,
                                     8.5450092830163271,
                                     25,
                                     10,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan')))
```

binduramarao, 2021-12-04:
Hello:

Really appreciate your help/ feedback.
Thanks,. wlil try it,
Bindu

closed 2021-12-06 by julianoes

binduramarao, 2021-12-06:
Thanks for promptly addressing my question on MissionItem issue in mission.py example.  Appreciate it, Bindu Rao