mavlink/MAVSDK-Python #415: How to connect Pixhawk to a Mac laptop and receive data using MAVSDK?

JinraeKim, 2021-12-06:
Hi, developers!

I'd like to know how to connect Pixhawk (4) to my M1 Macbook Air and receive data using MAVSDK?
I found an issue (#84), seems to be related to my situation, but `start_mavlink()` is not supported anymore.

I also tried to replace it by `await drone.connect(system_address="serial:///dev/ttyUSB0:57600")`, but it does not work.

Thanks in advance :)

JinraeKim, 2021-12-06:
I found the solution.
You can find a related USB modem (?), which is in `/dev`.
For example, in my case, the following change makes it work:
`await drone.connect(system_address="serial:///dev/tty.usbmodem01")`

julianoes, 2021-12-06:
Thanks for following up. Yes, that's right, for macOS. So this works now and the issue can be closed?

JinraeKim, 2021-12-06:
@julianoes Of course. I closed this issue with this comment.

closed 2021-12-06 by julianoes