mavlink/MAVSDK-Python #416: ImportError mavsdk-0.22.0

Jai-GAY, 2021-12-08:
Traceback (most recent call last):
  File "arm_quad_mavsdk.py", line 4, in <module>
    from mavsdk import System
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/__init__.py", line 7, in <module>
    from .system import System
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 6, in <module>
    from .async_plugin_manager import AsyncPluginManager
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/async_plugin_manager.py", line 3, in <module>
    import aiogrpc
  File "/home/pi/.local/lib/python3.7/site-packages/aiogrpc/__init__.py", line 1, in <module>
    from grpc import ChannelConnectivity, StatusCode, RpcError, RpcContext,\
  File "/home/pi/.local/lib/python3.7/site-packages/grpc/__init__.py", line 22, in <module>
    from grpc import _compression
  File "/home/pi/.local/lib/python3.7/site-packages/grpc/_compression.py", line 15, in <module>
    from grpc._cython import cygrpc
**ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by /home/pi/.local/lib/python3.7/site-packages/grpc/_cython/cygrpc.cpython-37m-arm-linux-gnueabihf.so)**

Python 3.7.3
pip 18.1 from /usr/lib/python3/dist-packages/pip (python 3.7)
$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

julianoes, 2021-12-08:
Can you try to update `grpc` installed via pip?

```
pip3 install --user --upgrade grpc
```

Which Raspberry Pi is this?

Jai-GAY, 2021-12-08:
> Which Raspberry Pi is this?

Pi4B

okay, later or tomorrow I try

julianoes, 2021-12-08:
https://groups.google.com/g/grpc-io/c/T91EyO81c8I

Jai-GAY, 2021-12-08:
> https://groups.google.com/g/grpc-io/c/T91EyO81c8I

ldd --version
ldd (Debian GLIBC 2.28-10+rpt2+rpi1) 2.28
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

likely no luck for pip3 install --user --upgrade grpc

Jai-GAY, 2021-12-08:
> pip3 install --user --upgrade grpc

$ pip3 install --user --upgrade grpc
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting grpc
  Downloading https://files.pythonhosted.org/packages/af/e6/59c8758fbb536a430c4409fba4098017d463a408858099e86ec1c96ab8fe/grpc-1.0.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-frbx0r22/grpc/setup.py", line 33, in <module>
        raise RuntimeError(HINT)
    RuntimeError: Please install the official package with: pip install grpcio

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-frbx0r22/grpc/

julianoes, 2022-03-24:
@ferglezr1998 thanks for the note. Which RPi and which RPi OS was that?

ferglezr1998, 2022-03-25:
@julianoes  I'm using Rpi 4B with Raspbian 10. And I have glibc 2.28 instead 2.29 that was required by grpc

julianoes, 2022-03-28:
Ok thanks.

I think in the past it worked well when I used python grpc as shipped by deb, I think `sudo apt install python3-grpcio`.

closed 2022-04-05 by julianoes

Jai-GAY, 2022-04-26:
Thank you so much for the pointers and apologies for the delay in responding.

In Pi Buster, as suggested, I uninstalled 1.42 and downgraded it to 1.40.0 for Python 3.7.
Now, I am[ having other issues](https://github.com/mavlink/MAVSDK-Python/issues/453#issuecomment-1108217843).
Interestingly, this does not happen in Pi Bullseye Pythom 3.9 with 1.43.0

sudo pip3 install grpcio==1.40.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting grpcio==1.40.0
  Downloading https://files.pythonhosted.org/packages/3f/c0/2922ed9b88adf7bea8d0d12870a3528f043ed90d1a6d0de6fde6cf3f5826/grpcio-1.40.0-cp37-cp37m-linux_armv7l.whl (38.4MB)
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3/dist-packages (from grpcio==1.40.0) (1.12.0)
Installing collected packages: grpcio
  Found existing installation: grpcio 1.16.1
    Not uninstalling grpcio at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'grpcio'. No files were found to uninstall.
Successfully installed grpcio-1.40.0

Name: grpcio
Version: 1.43.0
Summary: HTTP/2-based RPC framework
Home-page: https://grpc.io
Author: The gRPC Authors
Author-email: grpc-io@googlegroups.com
License: Apache License 2.0
Location: /home/pi/.local/lib/python3.9/site-packages
Requires: six
Required-by: mavsdk, aiogrpc