mavlink/MAVSDK-Python #418: pip3 installation got stuck on Raspberry Pi 4

JinraeKim, 2021-12-13:
```
pip3 install --user --upgrade mavsdk
Collecting mavsdk
  Using cached mavsdk-0.22.0-py3-none-manylinux2014_aarch64.whl (12.0 MB)
Collecting aiogrpc>=1.5
  Using cached aiogrpc-1.8-py3-none-any.whl (10 kB)
Collecting protobuf>=3.13.0
  Using cached protobuf-3.19.1-cp38-cp38-manylinux2014_aarch64.whl (913 kB)
Collecting grpcio>=1.11.0
  Using cached grpcio-1.42.0.tar.gz (21.3 MB)
Requirement already satisfied, skipping upgrade: six>=1.5.2 in /usr/lib/python3/dist-packages (from grpcio>=1.11.0->mavsdk) (1.14.0)
Building wheels for collected packages: grpcio
  Building wheel for grpcio (setup.py) ... -
```

Building wheel for grpcio seems to be on progress, but it takes too long time so I think it got stuck.

Does it typically take a long time, or is it a known issue?

JinraeKim, 2021-12-13:
One remedy: instead, I install `mavsdk` on docker in the build phase.

JinraeKim, 2021-12-13:
After waiting about 10 min, it is finally installed.
I think it's weird that it takes too long time.

JonasVautherin, 2021-12-13:
> Building wheel for grpcio (setup.py) ... -

It's building the `grpcio` package, which is a dependency of mavsdk-python. `grpcio` provides a binary for some platforms, but not all. It seems like `pip` decided that grpcio does not have a binary for your RPi4.

Which version of pip do you have? Did you try updating it?

julianoes, 2021-12-13:
You need to update pip like this:

```
pip3 install --upgrade pip
```

And then install mavsdk or grpcio like this:

```
python3 -m pip install mavsdk
```
```
python3 -m pip install grpcio
```

julianoes, 2021-12-13:
Exactly, I don't know exactly what the problem is.

closed 2021-12-13 by julianoes