mavlink/MAVSDK-Python #420: Sending RTCM messages from external RTK using mavsdk

CY-1992, 2021-12-14:
Hi,

I want to send RTCM messages to the drone px4 using mavsdk python. Mavlink passthrough  class is implemented in MAVSDK C++, but not in python wrapper. So I am trying to use pymavlink library to send RTCM in raw mavlink format. But I am already using the UDP port connection for mavsdk server.  So I can not use the same port for pymavlink to communicate (throws an error address is already in use).

```python
 await drone.connect(system_address= "udp://:14551")
```
```python
 mavutil.mavlink_connection('udpin:0.0.0.0:14551')
```
The above two lines would throws an error "address is already in use"

is there anyway we can send RTCM using the current mavsdk python implementation?  Are there any better ways to handle this situation?  or did I miss something? Any help would be appreciated.

Thanks

JonasVautherin, 2021-12-14:
The best would be to add the messages to MAVSDK :+1:. Would you be willing to contribute them?

CY-1992, 2021-12-15:
I really don't know much about mavsdk proto and templates. Having said that, I will try to look in to it. Please let me know the steps I need to do to implement mavlink passthrough in python wrapper.

JonasVautherin, 2021-12-15:
First step would be to define the interface you want in the proto files: https://github.com/mavlink/MAVSDK-Proto/tree/main/protos

julianoes, 2021-12-16:
See docs: https://mavsdk.mavlink.io/main/en/cpp/contributing/plugins.html#create-a-plugin

Katawann, 2022-02-08:
@CY-1992 there is an [existing issue](https://github.com/mavlink/MAVSDK/issues/112) open with the idea to implement RTK plugin in MAVSDK. Unfortunately I did not find the time to implement it yet, but this issue (and the related one) might give some info on how to implement it !

g1tm4n123, 2022-06-22:
@JonasVautherin this is very super
unfortunately, i can a little bit drone programming
i would like you to show me an example of the use of mavsdk RTK plugin
i connected to the vehicle on serial port and i have an ntrip server
await PXvhc.connect(system_address="serial:///dev/ttyUSB0:...")
ntrip caster xxx.xxx.xxx.xx
ntrip port 2101
ntrip mountpoint blox
ntrip username user
ntrip password pwd

Vehicle: Pixhawk
RTK rover: F9P

 _setup_stub(self, channel)
channel?
send_rtcm_data(self, rtcm_data)
from ntrim to rtcm_data?

PLS!

JonasVautherin, 2022-06-22:
@g1tm4n123 can you please open a new issue with your question? This one was a feature request which, I believe, can be closed now (because it got implemented).

closed 2022-06-22 by JonasVautherin