mavlink/MAVSDK-Python #422: connecting via TCP

Ricky-Millar, 2021-12-20:
hey, I think i've got a fundamental misunderstanding in how to connect to my drone, I'm trying to connect to mavlink through my tango2 tx's wifi.

I can get mission planner to connect using  TCP, 192.168.4.1 port: 5760, But within MAVSDK 
`drone.connect(system_address="tcp://[192.168.4.1][:5760]")`
Doesn't seem to connect. I assumed these would be the same as the address you would use for mission planner. 

I have code working within jmavsim all fine, just struggling to transfer it over to the real world.

julianoes, 2021-12-20:
The TCP connection is mostly untested because PX4 does not generally use TCP. What sort of radio are you using that gives you TCP? 

Can you get the output of mavsdk_server, e.g. by using:
```
logging.basicConfig(stream=sys.stdout, level=logging.DEBUG)
```

Or by starting mavsdk_server separately like this:
```
drone = System()
await drone.connect()
```
and then starting the `mavsdk_server` separately in another terminal. It is usually located in:
```
~/.local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server
```
(replace with your Python version)

Ricky-Millar, 2021-12-20:
ah alright, ill have a play around with that, thanks. 

It's the TBS tango 2. their TCP seems to work substantially better than UDP (i get like 95% lost packets with UDP, I've heard it's some problem with the radio), but I'm honestly not sure why its that way, it's just what they recommend using at the moment.

Ricky-Millar, 2021-12-20:
alright, the first method returns "None" 

for the second method, I'm not sure how you start the server from their terminal, sorry I'm not the most experienced programmer!

> The TCP connection is mostly untested because PX4 does not generally use TCP. What sort of radio are you using that gives you TCP?
>
> Can you get the output of mavsdk_server, e.g. by using:
>
> ``` [...]

julianoes, 2021-12-20:
For the second method, remove the arguments in `connect()` like above.

And then open a separate console and run mavsdk_server like this:
```
~/.local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server tcp://192.168.4.1:5760
```

To find where your `mavsdk_server` binary is, you can also look for it, using:
```
find ~ -name mavsdk_server
```

Ricky-Millar, 2021-12-20:
no matter if I try to connect first or not, assuming I've typed the right stuff into the console... I'm getting this error

`E1220 19:22:27.340000000  7244 server_chttp2.cc:50] {"created":"@1639981347.340000000","description":"No address added out of total 1 resolved","file":"D:\a\MAVSDK\MAVSDK\build\release\third_party\grpc\grpc\src\grpc\src\core\ext\transport\chttp2\server\chttp2_server.cc","file_line":587,"referenced_errors":[{"created":"@1639981347.340000000","description":"Failed to add port to server","file":"D:\a\MAVSDK\MAVSDK\build\release\third_party\grpc\grpc\src\grpc\src\core\lib\iomgr\tcp_server_windows.cc","file_line":512,"referenced_errors":[{"created":"@1639981347.340000000","description":"OS Error","file":"D:\a\MAVSDK\MAVSDK\build\release\third_party\grpc\grpc\src\grpc\src\core\lib\iomgr\tcp_server_windows.cc","file_line":208,"os_error":"Only one usage of each socket address (protocol/network address/port) is normally permitted.\r\n","syscall":"bind","wsa_error":10048}]}]}
Segmentation fault
`

julianoes, 2021-12-20:
Aha you're on Windows. That's good to know. It looks like it fails to bind. Do you have Mission Planner running when you try this?

And btw. I assume you're using ArduPilot and not PX4? Note that support for ArduPilot in MAVSDK is only experimental, you can see more details in https://github.com/mavlink/MAVSDK/issues/1568.

Ricky-Millar, 2021-12-20:
Ah yes, I should have said all my hardware at the start. I didn't think it could have an effect as I thought the incoming stream of mavlink data would be the same.

So I'm trying out a fairly different set up. It's a betaflight fc, that is transmitting mavlink over crossfire to my tx, and then tcp from my tx to my laptop/phone

I'm trying to make a gps-based antenna tracker that works with betaflight. As I said I have got it working with mission planner, so the mavlink is making it through to my PC/phone, and the rest of my code works in jmavsim too. My end game was to try and implement this onto a microcontroller or a raspi or something later on once I've got my brain around getting it working on the PC.

julianoes, 2021-12-20:
Ok cool!

Have you tried connecting without Mission Planner open? I think the problem is that Mission Planner is using the port which means MAVSDK cannot bind to it.

JonasVautherin, 2021-12-20:
I don't think it's a binding error, because here MAVSDK is a TCP client. It's not binding to anything, it's just opening a connection to a remote server (the drone that listens on 5760). Or did I miss something?

Just to be sure:

```python
drone.connect(system_address="tcp://[192.168.4.1][:5760]")
```

This is invalid, right? The way to write that is:

```python
drone.connect(system_address="tcp://192.168.4.1:5760")
```

Can you show us the command line you type and the corresponding output of `~/.local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server tcp://192.168.4.1:5760`?

Because you are on Windows, I assume you did not copy-paste exactly that, so it would be interesting for us to see exactly what you wrote.

Still guessing, but the error message sounds like grpc gets an error saying "Failed to add port to server" and "Only one usage of each socket address (protocol/network address/port) is normally permitted". But to me this is happening when `mavsdk_server` creates the grpc server, not when it tries to connect to the remote drone. So not related to `tcp://192.168.4.1:5760` but rather to `mavsdk_server` binding to 50051. Could you try a few different ports? With @julianoes' line above, it would be something like:

```
~/.local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 55555 tcp://192.168.4.1:5760
```

Where `-p 55555` tells `mavsdk_server` to listen on 55555 instead of 50051. Maybe 50051 is already used on your computer for some reason.

Ricky-Millar, 2021-12-20:
Oh sorry, I totally forgot to reply to that bit.
I intentionally disconnected from mission planner before trying to connect

I'll just go and make sure it isn't running in the back ground or anything and try again..

Ricky-Millar, 2021-12-20:
ok so, im on windows, to get that previous error I typed 
`'D:/Users/ricky/OneDrive/Desktop/100\ days\ of\ python/dronegpsmavlink/venv/Lib/site-packages/mavsdk/bin/mavsdk_server tcp://192.168.4.1:5760'`
into a bash terminal.

as for the `[192.168.4.1][:5760]` vs `192.168.4.1:5760` i have tried them both, but I can type anything in there at the moment with the same results.

I just tried again double checking mission planner is not running and still, the same result.

**BUT**
port 55555 I think has worked! I get
`$ D:/Users/ricky/OneDrive/Desktop/100\ days\ of\ python/dronegpsmavlink/venv/Lib/site-packages/mavsdk/bin/mavsdk_server -p 55555 tcp://192.168.4.1:5760
[08:44:49|Info ] MAVSDK version: v0.41.0 (D:\a\MAVSDK\MAVSDK\src\core\mavsdk_impl.cpp:21)
[08:44:49|Info ] Server started (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:48)
[08:44:49|Info ] Server set to listen on 0.0.0.0:55555 (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:49)
[08:44:49|Info ] Waiting to discover system on tcp://192.168.4.1:5760... (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:21)
`
You guys are pretty clever ay, haha. Any hints on how I could implement this into code?

julianoes, 2021-12-20:
Ah, so the mavsdk_server port is already taken. Now the question is what is using 50051. You could try to reboot the computer to make sure whatever is using is, is shut down.

Ricky-Millar, 2021-12-20:
got some more info for you, which I don't understand but maybe you do. I checked what is using the port 50051 in powershell too see what was blocking it, and it was mavask_server. 
I killed macask_server in task manager it started to work with the normal port.it works as long as I'm using  
`drone.connect()`

But If I try to use `drone.connect(system_address="tcp://192.168.4.1:5760")` then is stopps working again. I can see mavsdk still using the port in powershell but its not in the task manager for me to kill anymore. I'm struggling to make this repeatable and am a bit out of my depth here.

julianoes, 2021-12-20:
Good night over to NZ :wave: 

> I killed mavsdk_server in task manager and the bash command started to work with the normal port. it works as long as I'm using
drone.connect()

So it connects and everything is working? And if not, please paste the output that it's printing so we can debug what is going on.

If you use `drone.connect(system_address="tcp://192.168.4.1:5760")` with the arguments, then it will start mavsdk_server in the background, so you can only do either or.

Ricky-Millar, 2021-12-20:
Allright, I've not done much poking around to check if I've done something silly here, but this is what I get when the port works:
`[09:23:08|Info ] MAVSDK version: v0.41.0 (D:\a\MAVSDK\MAVSDK\src\core\mavsdk_impl.cpp:21)
[09:23:08|Info ] Server started (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:48)
[09:23:08|Info ] Server set to listen on 0.0.0.0:55550 (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:49)
[09:23:08|Info ] Waiting to discover system on tcp://192.168.4.1:5760... (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:21)
[09:23:38|Error] TCP receive error, trying to reconnect... (D:\a\MAVSDK\MAVSDK\src\core\tcp_connection.cpp:190)
[09:23:49|Error] TCP receive error, trying to reconnect... (D:\a\MAVSDK\MAVSDK\src\core\tcp_connection.cpp:190)
[09:24:11|Error] connect error: 10060 (D:\a\MAVSDK\MAVSDK\src\core\tcp_connection.cpp:87)
[09:24:11|Error] TCP receive error, trying to reconnect... (D:\a\MAVSDK\MAVSDK\src\core\tcp_connection.cpp:190)
[09:24:33|Error] connect error: 10060 (D:\a\MAVSDK\MAVSDK\src\core\tcp_connection.cpp:87)
[09:24:33|Error] TCP receive error, trying to reconnect... (D:\a\MAVSDK\MAVSDK\src\core\tcp_connection.cpp:190)
[09:24:55|Error] connect error: 10060 (D:\a\MAVSDK\MAVSDK\src\core\tcp_connection.cpp:87)
...`
etc....

alright, goodnight from me. maybe catch you again in 8-10 hours! haha, thanks for the help, you guys have been great.

Ricky-Millar, 2021-12-21:
so I've been trying a few things today, one of which is just switching to UDP. I've been trying that and also just been working with this example code to rule out my code is the issue:
 https://github.com/mavlink/MAVSDK-Python/blob/92ed8907d2a65245d15961c2e728c00c84ba2838/examples/telemetry.py
 
but I using `await drone.connect(system_address="tcp://192.168.4.1:5760")` 

but no matter what I'm still having the same issue as above. I saw mavsdk is for mavlink 2.0. I can't find any info about what version I'm using so ill send off an email to try and work that out!

julianoes, 2021-12-21:
Hm, that's interesting:
```
TCP receive error, trying to reconnect...
connect error: 10060
```

This means:
https://docs.microsoft.com/en-us/troubleshoot/windows-client/networking/10060-connection-timed-out-with-proxy-server

Have you made sure mavsdk_server gets to connect through the firewall? Sometimes that pops up on Windows...

And can you ping that IP? Does it respond?

Ricky-Millar, 2021-12-21:
A normal ping to 192.168.4.1 works fine, but in powershell if I use the Test-NetConnection thing to try ping the actual port (I tried with 3 or 4 ports) I get this:

`PS C:\Users\ricky> Test-NetConnection 192.168.4.1 -p 50051                                                              WARNING: TCP connect to (192.168.4.1 : 50051) failed                                                                                                                                                                                                                                                                                                                    ComputerName           : 192.168.4.1
RemoteAddress          : 192.168.4.1
RemotePort             : 50051
InterfaceAlias         : Wi-Fi
SourceAddress          : 192.168.4.2
PingSucceeded          : True
PingReplyDetails (RTT) : 2 ms
TcpTestSucceeded       : False
`

Again, this is outside of my understanding here so I don't know if that is valuable info, but seems like it pings but cant do TCP stuff.

Another thing I've been playing with is, my TX can either act as an access point that I connect to, or it can join the same network as my laptop. Is mavsdk-server able to work in one of those setups but not the other? 

I had a look at trying to do that regedit stuff, I don't know if windows 11 has changed regedit, but those instructions don't seem to relate to what my regedit looks like on this computer...

Ricky-Millar, 2021-12-22:
Ok not sure what I changed as I've changed a lot of stuff. but now I just get this 
```
[12:24:30|Info ] MAVSDK version: v0.41.0 (D:\a\MAVSDK\MAVSDK\src\core\mavsdk_impl.cpp:21)
[12:24:30|Info ] Server started (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:48
)
[12:24:30|Info ] Server set to listen on 0.0.0.0:5555 (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\
src\grpc_server.cpp:49)
[12:24:30|Info ] Waiting to discover system on tcp://192.168.4.3:5760... (D:\a\MAVSDK\MAVSDK
\src\mavsdk_server\src\connection_initiator.h:21)

ricky@Jumbotron ~
$ D:/Users/ricky/OneDrive/Desktop/100\ days\ of\ python/dronegpsmavlink/venv/Lib/site-packag
es/mavsdk/bin/mavsdk_server -p 5555 tcp://192.168.4.1:5760
[12:24:47|Info ] MAVSDK version: v0.41.0 (D:\a\MAVSDK\MAVSDK\src\core\mavsdk_impl.cpp:21)
[12:24:47|Info ] Server started (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:48
)
[12:24:47|Info ] Server set to listen on 0.0.0.0:5555 (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\
src\grpc_server.cpp:49)
[12:24:47|Info ] Waiting to discover system on tcp://192.168.4.1:5760... (D:\a\MAVSDK\MAVSDK
\src\mavsdk_server\src\connection_initiator.h:21)
[12:25:17|Error] TCP receive error, trying to reconnect... (D:\a\MAVSDK\MAVSDK\src\core\tcp_
connection.cpp:190)
[12:25:50|Error] TCP receive error, trying to reconnect... (D:\a\MAVSDK\MAVSDK\src\core\tcp_
connection.cpp:190)
[12:26:21|Error] TCP receive error, trying to reconnect... (D:\a\MAVSDK\MAVSDK\src\core\tcp_
connection.cpp:190)

```

julianoes, 2021-12-22:
It listens on port 5555 for gRPC so that the Python wrapper can talk to it, that's all correct.

> [12:24:47|Info ] Waiting to discover system on tcp://192.168.4.1:5760... (D:\a\MAVSDK\MAVSDK

Now for this part, it might be that the TCP implementation is "in the wrong direction". Basically, the TCP connection is the client part with:

https://github.com/mavlink/MAVSDK/blob/91eea12f7fa2fa7bb66c8ccd53b18d736141bfca/src/mavsdk/core/tcp_connection.cpp#L86

The other side would then have to implement the server and implement the `bind()`, `listen()`, `accept()` part.

Do you know what the other side implements? Maybe we have to add the server mode to mavsdk to support your case.

Ricky-Millar, 2021-12-22:
IF I sent you a Wireshark log of mission planner talking to the tx would you be able to tell from that? if not I can ask the manufacturers of the radio, they are quite quick to reply!

julianoes, 2021-12-22:
Either information might work, yes!

Ricky-Millar, 2021-12-22:
So I got a reply from tbs, (the tx manufacturer) and they say the tx is acting as a tcp server!

JonasVautherin, 2021-12-22:
> when I ask it to listen to 192.168.4.1

You are not asking to listen on 192.168.4.1. You are asking MAVSDK to open a TCP connection to 192.168.4.1:5760. So 192.168.4.1 is the IP of the drone, and MAVSDK will try to open a connection. Is that what you expect?

Ricky-Millar, 2021-12-23:
To be honest, at this point this has gone way deeper into network stuff than I understand. I'm trying to keep up with some googling but Im never sure what I expect, I'm learning on-the-go! the rest of this project with more hardware-based problems I'm alright with, just this wifi stuff I'm very new to.

julianoes, 2021-12-23:
> So I got a reply from tbs, (the tx manufacturer) and they say the tx is acting as a tcp server!

Ok, so this would then actually be correct. However, it doesn't work :thinking:.

It's a bit tricky to help you because TCP is not used much (at all?) with MAVSDK, at least when using PX4, and additionally Windows is not used much with it, so there are a few unknowns. Plus, I don't have the device.

A few things you could try:
- Try in WSL2.
- Try on a Linux box, RPi or something.
- Look at Wireshark to see what is going on.
- Start with environment variable `MAVSDK_MESSAGE_DEBUGGING` set to 1. That should give you potential console output of messages arriving.
- Try to connect with QGroundControl and check if that works.
- Try to connect with mavlink-router (not sure if that one actually works on Windows, probably not).

Ricky-Millar, 2021-12-24:
Nice one, well thanks again for all the help. I'll give this all a go when I'm back from Christmas holidays.

Have a good one!

julianoes, 2021-12-27:
You too, thanks!