mavlink/MAVSDK-Python #423: Why doesn't the second Command of「offboard.sset_position_ned」work?

moshibe, 2021-12-27:
I tried drone control by Offboard.
Command of「offboard.sset_position_ned」 didn't.
There was no error to give command, but the ｄrone was not moving.
Finally, the dorone landed successfully.
Why doesn't the second Command of「offboard.sset_position_ned」work?
※A guard timer of 10 seconds is set between commands.

①　drone.action.arm()
➁　drone.action.takeoff()　
➂　drone.offboard.set_position_ned()
④　drone.offboard.set_position_ned()
⑤　drone.action.land()

 please tell me how to do this?

julianoes, 2021-12-27:
It's hard or impossible to say based on this information, I'm afraid. In order to understand what is happening I would need to see the code that is running as well as a logfile of the flight. You can upload the log to logs.px4.io and then link it here.

moshibe, 2021-12-27:
Thank you for your answer. 
I am sorry for the insufficient information. 
We are using SITL Simulator (Software in the Loop). 
The code is compressed in a zip file. 
Thank you for your cooperation.

[test02.zip](https://github.com/mavlink/MAVSDK-Python/files/7779516/test02.zip)

moshibe, 2021-12-27:
Thank you for your answer. 
I'm sorry, I don't have enough information. 
I an using SITL Simulator (Software in the Loop). 
I have attached the source code in compressed form. 
Thank you very much for your help.

 [test02.zip](https://github.com/mavlink/MAVSDK-Python/files/7779615/test02.zip)

julianoes, 2021-12-27:
Can you create a logfile of when it does not work? The logs are saved in `build/px4_sitl_default/tmp/rootfs/log/` of the PX4 directory.

moshibe, 2021-12-28:
Thank you for your answer. 
I can't find the folder you specified. 
I found the logs in the following folder, which is attached. 

C:\ardupilot\ArduCopter\mav.tlog 
C:\ardupilot\ArduCopter\logs\LASTLOG.TXT
[log.zip](https://github.com/mavlink/MAVSDK-Python/files/7782831/log.zip)

moshibe, 2021-12-28:
My environment
PC Windows10 
simulator SITL Simulator (Software in the Loop) 
mavsdk 0.22.0.

moshibe, 2021-12-28:
My goal is to move the drone. 
Is it possible to use another command to move the drone instead of this "offboard.set_position_ned"?

julianoes, 2021-12-28:
Ok, it looks like you are using ArduPilot. Note that MAVSDK does not support ArduPilot fully. For a list of what is supposed to work, check https://github.com/mavlink/MAVSDK/issues/1568.

Also, you're version 0.22.0 is very old, make sure to use the latest one.

julianoes, 2021-12-28:
As you can see in https://github.com/mavlink/MAVSDK/issues/1568, offboard/guided support is not yet supported in MAVSDK for ArduPilot. You have to either debug it yourself or use something else like e.g. DroneKit.

moshibe, 2021-12-28:
Thank you for your answer. 
I understood that the MAVSDK does not support ArduPilot. 

Please let me know about dronekit-android. 
I am not able to find the documentation of the API in the following URL. 
https://github.com/dronekit/dronekit-android
Do you know, where is the documentation of the API?

julianoes, 2021-12-28:
I don't know. I'd have to google it as well.

closed 2021-12-28 by julianoes