mavlink/MAVSDK-Python #428: Can we inject actuator command via MAVSDK for offboard control?

JinraeKim, 2022-01-05:
I cannot find an example of injecting actuator commands for offboard control.

For example, thrust command or rotor angular rate.

julianoes, 2022-01-05:
What about these?

Actuator control: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/offboard.html#mavsdk.offboard.Offboard.set_actuator_control

Attitude rate control: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/offboard.html#mavsdk.offboard.Offboard.set_attitude_rate

JinraeKim, 2022-01-05:
Ok, there are appropriate functions :)

But there is no example with direct actuator control, right?

julianoes, 2022-01-05:
There is not but it would be great if you contribute one.

JinraeKim, 2022-01-05:
As I'm still not familiar with details of offboard control, I would appreciate if MAVSDK-Python provides an example of direct actuator control :)

JinraeKim, 2022-01-05:
> There is not but it would be great if you contribute one. 😄

Of course I will if I find an appropriate working example!

julianoes, 2022-01-05:
I think you can try it in simulation.

So do you actually need attitude rate control, or actuator control? Because that's not the same thing.

JinraeKim, 2022-01-05:
> I think you can try it in simulation.
> 
> So do you actually need attitude rate control, or actuator control? Because that's not the same thing.

Sorry for the confusion.

What I'm trying to do is actuator control.

julianoes, 2022-01-05:
Ok, and that's with PX4? And what is the use case? If I understand more context, I might be able to help with your actual problem.

JinraeKim, 2022-01-05:
> Ok, and that's with PX4? And what is the use case? If I understand more context, I might be able to help with your actual problem.

Yup, with PX4.

I'm trying to inject actuator control input from an external device (e.g., RPi or GCS) to Pixhawk (equipped with PX4) via MAVSDK.
This is for testing a custom controller and ease of usage.

JinraeKim, 2022-01-05:
BTW, what does "actuator control" stand for, exactly? (rotor thrust?)

julianoes, 2022-01-05:
Throttle in this case just means that the motor spins from low to max, 0..1, from what I can read.

JinraeKim, 2022-01-05:
> Throttle in this case just means that the motor spins from low to max, 0..1, from what I can read.

Thanks as always :)
So it's literally "throttle" like the throttle in RC controller.
Is there any systematic or heuristic way to send a thrust command like "the first rotor should be 1N"?

julianoes, 2022-01-05:
No, it's throttle as in PWM to a motor, I believe.

julianoes, 2022-01-05:
Just try it out in SITL ([with lockstep disabled](https://docs.px4.io/master/en/simulation/#disable-lockstep-simulation)).

JinraeKim, 2022-01-05:
Thank you a lot!

Traveller30, 2022-08-30:
@JinraeKim 
did it worked?
can you send direct pwm cmds to motors from offboard by mavsdk?

JinraeKim, 2022-08-30:
> @JinraeKim 
> did it worked?
> can you send direct pwm cmds to motors from offboard by mavsdk?
> 
> 

TBH, I don't remember clearly cuz our team is decided not to use MAVSDK for offboard control... I'm sorry.
I guess it didn't work well (or was hard to figure out what each variable means).

@hnlee77 Do you remember any things about this?

hnlee77, 2022-08-31:
> TBH, I don't remember clearly cuz our team is decided not to use MAVSDK for offboard control... I'm sorry. I guess it didn't work well (or was hard to figure out what each variable means).
> 
> @hnlee77 Do you remember any things about this?

It didn't work well. we couldn't send direct pwm cmds to motors.
so now, we decided send only attitude cmds for offboard by mavsdk.

JonasVautherin, 2022-09-01:
Kind reminder: contributions are always welcome if a MAVSDK feature doesn't work well :+1:.

rakmo97, 2023-06-12:
Hello, @JinraeKim , just checking to see if there are any updates on this? I'm trying to send a command directly to each quadcopter motor rather than use higher level functions like "takeoff" or "land", and an example would be very helpful. Have you figured it out?

JinraeKim, 2023-06-12:
> Hello, @JinraeKim , just checking to see if there are any updates on this? I'm trying to send a command directly to each quadcopter motor rather than use higher level functions like "takeoff" or "land", and an example would be very helpful. Have you figured it out?

We are no longer working on this topic so there is no update so far :<

julianoes, 2023-06-12:
Offboard control using actuator controls has actually been removed in more recent versions of PX4.

rakmo97, 2023-06-12:
Thanks! Does that mean the offboard.set_actuator_control function wouldn't work?

I know this might not be the place for this question, but how are people sending actuator commands from a host computer to their quadcopter? I've been trying to figure that out using mavsdk-python.

julianoes, 2023-06-13:
It should still work for v1.13: https://github.com/PX4/PX4-Autopilot/blob/release/1.13/src/modules/mavlink/mavlink_receiver.cpp#LL160C3-L160C45