mavlink/MAVSDK-Python #432: system.telemetry updates not firing?

m-sienk, 2022-01-15:
If I run my code and then start Gazebo simulator only the connection state value will update (check_connection_state()). All other functions will exit as the there is never an initial value returned by the telemetry plugin (see output).

Is this intended functionality?

**main.py**
```
import asyncio
from aircraft import Aircraft

async def main():
    aircraft = Aircraft()
    await aircraft.connect()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    try:
        loop.create_task(main())
        loop.run_forever()
        #loop.run_until_complete(main())
    finally:
        loop.stop()
        loop.close()
```

**aircraft.py**
```
import asyncio
from mavsdk import System

class Aircraft:

    def __init__(self):
        print("Aircraft init")
        self.system = System()

    async def connect(self):
        await self.system.connect(system_address="udp://:14540")
        await self.start_status_update()

    async def start_status_update(self):
        asyncio.create_task(self.check_connection_state())
        asyncio.create_task(self.get_flight_mode_status())
        asyncio.create_task(self.get_armed_status())
        asyncio.create_task(self.get_health_all_ok_status())

    async def check_connection_state(self):
        async for connection_state in self.system.core.connection_state():
            print("connection_state update")
        print("-- exit check connection state loop --")

    async def get_flight_mode_status(self):
        async for flight_mode in self.system.telemetry.flight_mode():
            print("flight_mode update")
        print("-- exit flight mode loop --")    

    async def get_armed_status(self):
        async for armed in self.system.telemetry.armed():
            print("armed update")
        print("-- exit armed loop --")

    async def get_health_all_ok_status(self):
        async for health_all_ok in self.system.telemetry.health_all_ok():
            print("health ok update")
        print("-- exit health ok loop --")
```

**output:**
```
Aircraft init
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
-- exit flight mode loop --
-- exit armed loop --
-- exit health ok loop --
connection_state update
```

JonasVautherin, 2022-01-15:
Where do you start your event loop? Did you try to run one of the examples? https://github.com/mavlink/MAVSDK-Python/tree/main/examples

m-sienk, 2022-01-15:
Thanks for your reply.

I put this example together to highlight my point its just the two files I've included, ​the event loop starts in main.py.

```
if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    try:
        loop.create_task(main())
        loop.run_forever()
        #loop.run_until_complete(main())
    finally:
        loop.stop()
        loop.close()
```

I have run the examples in https://github.com/mavlink/MAVSDK-Python/tree/main/examples and I'm able to receive a telemetry stream from a simulated device in Gazebo simulator. The code I have included works as expected provided you start Gazebo sim and then run the code, reverse the order and you get the output.

```
Aircraft init
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
-- exit flight mode loop --
-- exit armed loop --
-- exit health ok loop --
connection_state update
```

I'm trying to set this up so each function in aircraft.py - start_status_update, check_connection_state ... etc will each run forever as a coroutine. However every function except check_connection_state exits (as displayed in the output).

[example.zip](https://github.com/mavlink/MAVSDK-Python/files/7876027/example.zip)

JonasVautherin, 2022-01-15:
> The code I have included works as expected provided you start Gazebo sim and then run the code, reverse the order and you get the output.

Does the reverse order work for the examples?

m-sienk, 2022-01-15:
If you run https://github.com/mavlink/MAVSDK-Python/blob/main/examples/telemetry.py then start Gazebo you get the following output:

```
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
```

then no updates.

JonasVautherin, 2022-01-16:
I just tried to install mavsdk from pip (`pip install --user mavsdk`) and run the following four scenarios, which work for me:

Takeoff example, SITL run after the script:

1. Run `python takeoff_and_land.py`
2. Run `docker run --rm -it -p 8554:8554 jonasvautherin/px4-gazebo-headless:1.12.1 -v typhoon_h480`

Takeoff example, SITL run before the script:

1. Run `docker run --rm -it -p 8554:8554 jonasvautherin/px4-gazebo-headless:1.12.1 -v typhoon_h480`
2. Run `python takeoff_and_land.py`

Telemetry example, SITL run after the script:

1. Run `python telemetry.py`
2. Run `docker run --rm -it -p 8554:8554 jonasvautherin/px4-gazebo-headless:1.12.1 -v typhoon_h480`

Telemetry example, SITL run before the script:

1. Run `docker run --rm -it -p 8554:8554 jonasvautherin/px4-gazebo-headless:1.12.1 -v typhoon_h480`
2. Run `python telemetry.py`

Would you mind trying them on your end?

m-sienk, 2022-03-01:
Sorry for the late reply. I tried this again with a fresh install and it now works as expected. Thanks for your help.

Although now I cant get the aircraft to takeoff in Gazebo unless I have QGroundControl running or I set the 'NAV_RCL_ACT' flag to 0.

closed 2022-03-01 by m-sienk