mavlink/MAVSDK-Python #433: how to control velocity of fw using mavsdk

airsimdevdd, 2022-01-24:
I want to limit the velocity of fixedwing at certain waypoints is it possible ? can anyone help me out .thanks in advance.

JonasVautherin, 2022-01-24:
Did you try setting the [speed](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/mission/mission.proto#L202) in the mission item?

airsimdevdd, 2022-01-25:
can we only speed in mission item because im using attitude setpoints for navigation

JonasVautherin, 2022-01-25:
Oh so you're using offboard mode? Did you try one of the [fw params](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#fw-tecs)?

airsimdevdd, 2022-01-25:
no use tried setting it

julianoes, 2022-01-27:
> can we only use or set speed in mission item because im using attitude setpoints for navigation

I don't understand what you mean. Are you using mission mode with waypoints or attitude setpoints?

airsimdevdd, 2022-01-27:
im using attitude setpoints

julianoes, 2022-01-27:
So this API? https://mavsdk.mavlink.io/main/en/cpp/api_reference/structmavsdk_1_1_offboard_1_1_attitude.html

airsimdevdd, 2022-01-27:
currently using mavsdk-python api offboard attitude

julianoes, 2022-01-28:
Ok, so with attitude control you have to control thrust which is probably suboptimal for fixedwing. I can see how that's not optimal. Let me ask in Slack if someone has a better suggestion.

airsimdevdd, 2022-02-09:
ok thank you waiting for your response