mavlink/MAVSDK-Python #434: is there any direct access to ground speed ?

MehdiRifi, 2022-01-28:
Hey everyone,
im trying to read the ground speed but can't find directly as the case when using dronekit with ardupilot, am i missing something ??

if not possible is there any way i can listen to this mavlink message and read the content [VFR_HUD]
(https://mavlink.io/en/messages/common.html#VFR_HUD) ?

if none above is possible, is there any formula to calculate it using the ned velocities? besides this one sqrt(north_ms*north_ms + east_ms*east_ms) because ive tried it and it gives wrong values .

thanks

julianoes, 2022-01-31:
> sqrt(north_msnorth_ms + east_mseast_ms)

This should be correct. Are you using PX4?

MehdiRifi, 2022-01-31:
you're right it's correct, ive re examined what ive done and the trick is that i was doing an async sleep inside the subscription method and somehow it print wrong values, cases:

1.  with the async sleep when starting the monitoring while the velocity is approximately zero => wrong values 
2.  with the async sleep when starting the monitoring while the velocity is different than zero (drone is moving) => correct values
3. without async sleep it's always up to date  => correct values

sorry my mistake, any insight about this behavior ?

julianoes, 2022-02-01:
Why would it be wrong at ~ 0? I don't understand that :thinking:

MehdiRifi, 2022-02-01:
```
async for velocity in drone.telemetry.velocity_ned():
          speed = math.sqrt(velocity.east_m_s * velocity.east_m_s + velocity.north_m_s * velocity.north_m_s)
          print(speed)
          await asyncio.sleep(0.5)
```
when executing with the sleep instruction and the drone is not moving it gives ~ 0 which's normal but when start moving it will stack at the same range of values with a minimum of change,
if u stop and restart again while the drone is still moving you will get the correct value but when the drone is stopped again it will stick to the last range of speed.

when removing the sleep instruction it works as expected.

PS: i don't have much experience with python so maybe i'm missing something which is normal to the python world.

julianoes, 2022-02-02:
Right, I think you're not supposed to use sleep in there.

Have a look at https://auterion.com/mavsdk-python-easy-asyncio/

closed 2022-02-02 by julianoes