mavlink/MAVSDK-Python #435: ImportError: cannot import name 'System' from partially initialized module 'mavsdk'

exabotixmpaul, 2022-02-03:
Hi, I'm trying to make my first attempts with mavsdk and installed the whole thing via `pip install mavsdk`, unfortunately I get an import error.

`ImportError: cannot import name 'System' from partially initialized module 'mavsdk'`

does anyone have an idea what might have gone wrong ?

exabotixmpaul, 2022-02-03:
under .local/lib/python3.9/site-packages/mavsdk is system.py included

JonasVautherin, 2022-02-03:
What did you try to do? Did you try to run `$ python` and type `from mavsdk import System`?

exabotixmpaul, 2022-02-03:
exactly

```
import json
import requests
import time
from datetime import datetime
import threading
import asyncio
from mavsdk import system

async def main():
    flightplan_src = "ufly1"
    flightplan_id = "1234"

    drone = System()
    await drone.connect(system_adress = "serial:///dev/ttyUSB0:57600")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("Drone discovered!")
            break

...
```

exabotixmpaul, 2022-02-03:
What an embarrassing mistake, yes I had called the file that quickly to try it out.

However, it now reports the next error

```
Traceback (most recent call last):
  File "/home/pi/json-parser/test.py", line 16, in <module>
    from mavsdk import System
  File "/home/pi/.local/lib/python3.9/site-packages/mavsdk/__init__.py", line 7, in <module>
    from .system import System
  File "/home/pi/.local/lib/python3.9/site-packages/mavsdk/system.py", line 6, in <module>
    from .async_plugin_manager import AsyncPluginManager
  File "/home/pi/.local/lib/python3.9/site-packages/mavsdk/async_plugin_manager.py", line 3, in <module>
    import aiogrpc
  File "/home/pi/.local/lib/python3.9/site-packages/aiogrpc/__init__.py", line 1, in <module>
    from grpc import ChannelConnectivity, StatusCode, RpcError, RpcContext,\
  File "/home/pi/.local/lib/python3.9/site-packages/grpc/__init__.py", line 22, in <module>
    from grpc import _compression
  File "/home/pi/.local/lib/python3.9/site-packages/grpc/_compression.py", line 15, in <module>
    from grpc._cython import cygrpc
ImportError: /home/pi/.local/lib/python3.9/site-packages/grpc/_cython/cygrpc.cpython-39-arm-linux-gnueabihf.so: undefined symbol: __atomic_exchange_8
```

JonasVautherin, 2022-02-03:
Good catch @MehdiRifi!

> However, it now reports the next error

Does it say "Waiting for drone to connect..." before the error? Does it say "Drone discovered!"? I want to know what throws that error.

exabotixmpaul, 2022-02-03:
no goes straight into this error

JonasVautherin, 2022-02-04:
Yeah it feels like an error with aiogrpc :thinking:

MehdiRifi, 2022-02-04:
yes i think there is a build problem at the grpc lib for some Raspberry Pi OS, take a look at this [undefined symbol: __atomic_exchange_8](https://stackoverflow.com/questions/70084134/undefined-symbol-atomic-exchange-8) and this [Fix libatomic linking on Raspberry Pi OS Bullseye #28041
](https://github.com/grpc/grpc/pull/28041)

exabotixmpaul, 2022-02-04:
i was able to fix it with

```
pip3 uninstall -y grpcio grpcio-tools
sudo apt install -y python3-grpcio python3-grpc-tools
```

thanks for your help @JonasVautherin and @MehdiRifi ;-)

closed 2022-02-04 by exabotixmpaul