mavlink/MAVSDK-Python #436: MAVSDK on Raspberry Pi 4 receiving error

brentonduvall, 2022-02-07:
Hello, I am currently using MAVSDK to try to connect my Pixhawk 4 flight controller through the telem 2 port two the TX and RX on the Pi. When I try "from mavsdk import System" I receive multiple errors such as "Import error: __atomic_exchange_8". Can I get some help or direction? Thank you.

JonasVautherin, 2022-02-07:
You would need to tell us how you installed it, which version, and show more of the debug output.

MehdiRifi, 2022-02-08:
try reinstalling grpc: 
pip3 uninstall -y grpcio grpcio-tools
sudo apt install -y python3-grpcio python3-grpc-tools

brentonduvall, 2022-02-09:
Hello,

The solution MehdiRifi suggested worked. Although we are running into some more issues regarding connecting mavsdk to the Pixhawk 4 flight controller. We have everything wired correctly, and is using the serial port on a raspberry pi to connect to it. When using the arm, takeoff, and land example code the code just stops. We are unsure of if mavsdk has connected to the drone and were wondering is there was any help we could get? This is for our capstone. More info can be provided if necessary. Thank you

julianoes, 2022-02-10:
@brentonduvall this sounds like a different issue.

Please create a new issue and copy in the output that you get from mavsdk_server. You can get it using these steps:
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

closed 2022-02-10 by julianoes