mavlink/MAVSDK-Python #437: Problems uploading raw mission items

mokidorne, 2022-02-09:
Hi it is unfortunately not possible for me to create a valid mission item with rawmission that I have uploaded

```
# -*- coding: utf-8 -*-

"""

exabotix GmbH
Autor: Moritz Paul

"""

import json
import requests
import asyncio
from mavsdk import System
from mavsdk.mission_raw import MissionItem

async def main():
    flightplan_src = "ufly1"
    flightplan_id = "1234"

    with open("cmd/flightplan.json", "r") as file:
        flightplan_cmd_json = json.load(file)

    drone = System()
    await drone.connect(system_address = "serial:///dev/ttyUSB0:57600")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("Drone discovered!")
            break

    await drone.mission_raw.clear_mission()
    mission_items = []

    for i in range(len(flightplan_cmd_json)):
        if flightplan_cmd_json["command"][i]["id"] == "waypoint":
            print(i+1)
            mission_items.append(MissionItem(1,#Seq
                                 6,#frame
                                 16,#command
                                 0,#current
                                 1,#autocontinue
                                 0,#param1
                                 10,#param2
                                 0,#param3
                                 float('nan'),#param4
                                 402603986,#x
								 -754579572,#y
								 50,#z
								 0))#MAV_MISSION_TYPE_MISSION
        """elif flightplan_cmd_json["command"][i]["id"] == "speed":
            mission_items.append(MissionItem(i+1,#Seq
                                 2,#frame
                                 178,#command
                                 0,#current
                                 1,#autocontinue
                                 1.0,#param1
                                 flightplan_cmd_json["command"][i]["speed"],#param2
                                 -1.0,#param3
                                 0.0,#param4
                                 0,#x
								 0,#y
								 0.0,#z
								 0))#MAV_MISSION_TYPE_MISSION
        elif flightplan_cmd_json["command"][i]["id"] == "land":
            mission_items.append(MissionItem(i+1,#Seq
                                 3,#frame
                                 21,#command
                                 0,#current
                                 1,#autocontinue
                                 0.0,#param1
                                 0.0,#param2
                                 0.0,#param3
                                 float('nan'),#param4
                                 int(flightplan_cmd_json["command"][i]["latitude"] * 10**7),#x
								 int(flightplan_cmd_json["command"][i]["longitude"] * 10**7),#y
								 float(flightplan_cmd_json["command"][i]["altitude"]),#z
								 0))#MAV_MISSION_TYPE_MISSION
        elif flightplan_cmd_json["command"][i]["id"] == "takeoff":
            mission_items.append(MissionItem(i+1,#Seq
                                 3,#frame
                                 22,#command
                                 0,#current
                                 1,#autocontinue
                                 0.0,#param1
                                 0.0,#param2
                                 0.0,#param3
                                 float('nan'),#param4
                                 int(flightplan_cmd_json["command"][i]["latitude"] * 10**7),#x
								 int(flightplan_cmd_json["command"][i]["longitude"] * 10**7),#y
								 float(flightplan_cmd_json["command"][i]["altitude"]),#z
								 0))#MAV_MISSION_TYPE_MISSION
        else:
            print("unknown command")"""

    print("-- Uploading mission")
    await drone.mission_raw.upload_mission(mission_items)

if __name__ == '__main__':
    asyncio.get_event_loop().run_until_complete(main())

```

I have now limited it to just waypoints with fixed values, but unfortunately without success with this as well.
The goal should be that I use the commands waypoint, takeoff, speed, land.

Output:
```
python upload_mission.py
Waiting for drone to connect...
Drone discovered!
3
-- Uploading mission
Traceback (most recent call last):
  File "/home/pi/json-parser/upload_mission.py", line 100, in <module>
    asyncio.get_event_loop().run_until_complete(main())
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/pi/json-parser/upload_mission.py", line 97, in main
    await drone.mission_raw.upload_mission(mission_items)
  File "/home/pi/.local/lib/python3.9/site-packages/mavsdk/mission_raw.py", line 711, in upload_mission
    raise MissionRawError(result, "upload_mission()", mission_items)
mavsdk.mission_raw.MissionRawError: INVALID_ARGUMENT: 'Invalid Argument'; origin: upload_mission(); params: ([<mavsdk.mission                            _raw.MissionItem object at 0xb500c388>],)

```

The JSON file I use currently only contains a waypoint at position 3.

MehdiRifi, 2022-02-09:
Hey please follow these rules:

1. All element must have **_seq_** number as an int starting from 0
2. One element of the mission items must be the current usally i set the first one => current must be set to 1
3. **Waypoint**, **RTL**, **LAND** ,**Takeoff** items must set **_param1,param2,param3_** to _**zero**_ and _**param4**_ to _**float('nan')**_
4. RTL item must have mission frame set to _**mavutil.mavlink.MAV_FRAME_MISSION**_ which translated to number **_2_**
5. x,y must be multiplied by **_10^7_** and be a valid int or long
6. Mission type (last param) must be set to _**mavutil.mavlink.MAV_MISSION_TYPE_MISSION**_
7. Change speed item must have frame set to _**mavutil.mavlink.MAV_FRAME_MISSION**_ which translated to number **_2_**
8. Change speed item must set
 **param1** for speed type
 **param2** for value
 **param3** to 0 
 **param4** set to float('nan')

Working example:

```
      MissionItem(
                              seq=0,
                              frame=mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT,
                              command=mavutil.mavlink.MAV_CMD_NAV_WAYPOINT,
                              current=1,
                              autocontinue=1,
                              param1=0,
                              param2=0,
                              param3=0,
                              param4=float('nan'),
                              x=lat * 10 ** 7,
                              y=lon * 10 ** 7,
                              z=alt,
                              mission_type=mavutil.mavlink.MAV_MISSION_TYPE_MISSION
                   )
```

mokidorne, 2022-02-09:
works with all 4 maneuvers thank you very much

closed 2022-02-09 by mokidorne

Ice-SleepingWater, 2023-05-29:
> works with all 4 maneuvers thank you very much

Hi Mokidorne. May I know how u make the mission_raw work with local x&y coordinate? Thank you.

julianoes, 2023-05-29:
FYI local coordinates are not supported in PX4.