mavlink/MAVSDK-Python #439: MAVSDK -- Import a QGroundControl missions in JSON .plan format.

StavrosKontzias29, 2022-02-10:
Hello,

First time working with MAVSDK. Made a flight mission on Q-Ground Control and saved it to the JSON .plan format and want to import it to MAVSDK to run it. I am having trouble on how to approach this. I took a look at the documentation for the MAVSDK-Python docs but am a little stuck here. Any help would be appreciated! 

Thank you

julianoes, 2022-02-10:
Try this: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/mission_raw.html#mavsdk.mission_raw.MissionRaw.import_qgroundcontrol_mission

StavrosKontzias29, 2022-02-10:
@julianoes Thank you for the suggestion.  Would it resemble something like below? Also what does it take to run the QGC mission plan from mavsdk?

    async def import_qgroundcontrol_mission(self, qgc_plan_path):
       
        qgc_plan_path = 'C:\\Users\\QGC_FLIGHT.plan'

        request = mission_raw_pb2.ImportQgroundcontrolMissionRequest()

        request.qgc_plan_path = qgc_plan_path

        response = await self._stub.ImportQgroundcontrolMission(request)

        result = self._extract_result(response)

        if result.result != MissionRawResult.Result.SUCCESS:
            raise MissionRawError(result, "import_qgroundcontrol_mission()", qgc_plan_path)

        return MissionImportData.translate_from_rpc(response.mission_import_data)

Thanks

closed 2022-02-11 by julianoes

julianoes, 2022-04-07:
I would say that's exactly something that Python async allows you to do, check the docs: https://docs.python.org/3/library/asyncio-task.html#coroutines

I tried it and came up with this, I think that's what you wanted:

```

task = asyncio.create_task(drone.mission_raw.upload_mission(mission_import_data.mission_items))
print("Mission uploading in background...")
# do something else here...
await task
print("Mission uploaded.")
```

@kennethakiti1 next time, please create a new issue so that it's easier to find them for others.

kennethakiti1, 2022-04-11:
sorry would do that thanks

SebastianViollaz, 2023-04-05:
Is there a way to import an strucutre scan .plan? i read in doc that it isn't supported, maybe it isn't updated and there's a feature to do it. If it isn't supported could be good to implement that.

igoutta, 2023-12-19:
Hi, I am wondering if you are aware that the import function it's _**NOT** currently working_ as excepted. 

I'm currently developing a computer companion module with your SDK and the problem is the following. The system is commanded to load a mission previously acquired through QGroundControl. The mission in question it's really simple, just a few points to go through. The problem arises at the moment the system is commanded to upload the mission, because the error shows the invalid argument description. I researched in this repo the details of this issue and the I encountered that such a problem usually happens when the **sequence is bad** and/or the current number is more than one in the entire mission. Then I printed the items processed by the SDK and then I found a double zero at the first two items, differents frames that really shouldn't be a problem, current value in the second item, and the param 4 is zero at the first item. The first item apparently is taken from the planned home position of the Qgroundcontrol plan which only has 3 values corresponded to the drone's 3D position. So I'm currently bypassing that first item but I'm looking forward to a solution in this matter. 

[image]

Qgroundcontrol mission plan: 
```
{
    "fileType": "Plan",
    "geoFence": {
        "circles": [
        ],
        "polygons": [
        ],
        "version": 2
    },
    "groundStation": "QGroundControl",
    "mission": {
        "cruiseSpeed": 1,
        "firmwareType": 3,
        "hoverSpeed": 5,
        "items": [
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 0,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 1,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    0,
                    -2.1157681120467093,
                    -79.9385202714547,
                    0
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 0,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 2,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    -2.1158699615280367,
                    -79.93851893061948,
                    0
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 0,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 3,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    -2.116022744157194,
                    -79.93852294714648,
                    0
                ],
                "type": "SimpleItem"
            }
        ],
        "plannedHomePosition": [
            -2.295260803823127,
            -79.7604224,
            0
        ],
        "vehicleType": 10,
        "version": 2
    },
    "rallyPoints": {
        "points": [
        ],
        "version": 2
    },
    "version": 1
}

```
Thanks for reading and sorry for the trouble.

julianoes, 2023-12-20:
@igoutta can you please create a new issue and give all the context for me to look into.

julianoes, 2023-12-20:
Nevermind, that's what you did.