mavlink/MAVSDK-Python #441: Altitude type in offboard Positionglobalyaw

airsimdevdd, 2022-02-13:
what is format for altitude in mavsdk offboard Positionglobalyaw how to define it and will it work properly can anyone letme know

julianoes, 2022-02-19:
Can you share some code what you're trying to do? I don't quite understand what you mean.

julianoes, 2022-03-02:
It's difficult to understand one line out of context. Please show a small example which demonstrates your problem or helps me reproduce it.

Moseskota, 2023-01-04:
> It's difficult to understand one line out of context. Please show a small example which demonstrates your problem or helps me reproduce it.

#!/usr/bin/env python3

import asyncio

from mavsdk import System
from mavsdk.offboard import (PositionGlobalYaw, OffboardError)

async def run():
    drone = System()
    await drone.connect(system_address="udp://:14540")
  # await drone.connect("serial:///dev/ttyACM0:57600")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok and health.is_home_position_ok:
            print("-- Global position estimate OK")
            break

    print("-- Arming")
    await drone.action.arm()

    print("---takeoff")
    await drone.action.takeoff()

    print("-- Setting initial setpoint")
    await drone.offboard.set_position_global(PositionGlobalYaw(0.0, 0.0,2.5, 0.0,PositionGlobalYaw.AltitudeType(0)))
    await asyncio.sleep(10)

    print("-- Starting offboard")
    try:
        await drone.offboard.start()
    except OffboardError as error:
        print(f"Starting offboard mode failed with error code: {error._result.result}")
        print("-- Disarming")
        await drone.action.disarm() 
        return

    # async def print_z_velocity(drone):
    #     async for odom in drone.telemetry.position_velocity_ned():
    #         print(f"{odom.velocity.north_m_s} {odom.velocity.down_m_s}")

   # asyncio.ensure_future(print_z_velocity(drone))

    print("-- Go 0m North, 0m East, -10m Down within local coordinate system")
    await drone.offboard.set_position_global(PositionGlobalYaw(0.0, 0.0, -100.0, 0.0, PositionGlobalYaw.AltitudeType(0)))

    await asyncio.sleep(10)

        
    print("-- Stopping offboard")
    try:
        await drone.offboard.stop()
    except OffboardError as error:
        print(f"Stopping offboard mode failed with error code: {error._result.result}")

    await drone.action.land()
   

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

Moseskota, 2023-01-04:
@julianoes I am facing something very similar when trying to execute a simple script with Offboard_PositionGlobalYaw. 
Waiting for drone to connect...
-- Connected to drone!
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Arming
---takeoff
-- Setting initial setpoint
-- Starting offboard
Starting offboard mode failed with error code: NO_SETPOINT_SET
-- Disarming
Traceback (most recent call last):
  File "COC_offboard_pos_global_yaw.py", line 38, in run
    await drone.offboard.start()
  File "/home/moses/anaconda3/lib/python3.8/site-packages/mavsdk/offboard.py", line 1150, in start
    raise OffboardError(result, "start()")
mavsdk.offboard.OffboardError: NO_SETPOINT_SET: 'No Setpoint Set'; origin: start(); params: ()

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "COC_offboard_pos_global_yaw.py", line 69, in <module>
    loop.run_until_complete(run())
 File "/home/moses/anaconda3/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "COC_offboard_pos_global_yaw.py", line 42, in run
    await drone.action.disarm() 
  File "/home/moses/anaconda3/lib/python3.8/site-packages/mavsdk/action.py", line 343, in disarm
    raise ActionError(result, "disarm()")
mavsdk.action.ActionError: COMMAND_DENIED_NOT_LANDED: 'Command Denied Not Landed'; origin: disarm(); params: ()

[image]
[image]

julianoes, 2023-01-04:
Why are you all trying to go to latitude and longitude 0? I assume there is a "bug" that prevents that but it's also an odd thing to do, unless you wan to fly to the [Null Island](https://en.wikipedia.org/wiki/Null_Island).

closed 2023-02-03 by airsimdevdd