mavlink/MAVSDK-Python #442: Pause Mission

AjayaDahal, 2022-02-13:
Hello,
I am working on a project to create an autonomous plane. An example file: mission.py is very helpful. I am stuck in one critical step and looking for some guidance. During the mission, I need to pause the current mission and fly to a different coordinate. Upon reaching to that coordinate, I need to continue the paused mission following the next way point. Thank you.

kennethakiti1, 2022-05-27:
> then to mission.start_mission() to resume?

@JonasVautherin  does that mean that after pausing a mission (which takes the drone into hold position), I can simply resume the mission from the current position (without having to restart)?

because I have been trying to figure out if there is a resume mission command but I cannot seem to find any. thanks

JonasVautherin, 2022-05-27:
If you look at what [`mission.start_mission()`](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission/mission_impl.cpp#L769-L772) does, you'll see that it "just" goes back to the mission mode. I believe that this resumes the mission.

If you want to restart the mission, I would say that you have to use [mission.set_current_mission_item](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/mission/mission.proto#L73-L82):

    /*
     * Sets the mission item index to go to.
     *
     * By setting the current index to 0, the mission is restarted from the beginning. If it is set
     * to a specific index of a mission item, the mission will be set to this item.
     *
     * Note that this is not necessarily true for general missions using MAVLink if loop counters
     * are used.
     */