mavlink/MAVSDK-Python #444: Can't go through waypoints on Fixed Wing

Erush01, 2022-02-20:
I'm trying to do a autonomous 8 figure with Fixed Wing using MAVSDK-Python with PX4. When I use goto_location() it loiters around that waypoint. How can I make it go through waypoints.
[image]
.

JonasVautherin, 2022-02-21:
Did you try planning a mission?

Erush01, 2022-02-21:
> 
![Screenshot from 2022-02-21 01-32-3
[image]
![Screenshot from 2022-02-20 13-01-23](https://user-im
[image]
ages.githubusercontent.com/69166608/154954668-0bb37b94-94b7-4e6c-8e0d-60e7bde6f5b5.png)

2](https://user-images.githubusercontent.com/69166608/154954568-393442bd-a9ab-40a7-9941-c9f85ef72a64.png)

I tried with both QGroundControl and from code. I think I did something wrong but I dont know what it is.

Erush01, 2022-02-21:
> Did you try planning a mission?

I tried to set a plan with both QGroundControl and from code. But both didnt worked like as I expected. 

[image]

[image]

[image]

[image]

JonasVautherin, 2022-02-21:
I guess the fixed wing has some limitations on the curves it can fly, and depending on how the waypoints are set, it won't be able to fly directly from one to the other.

If you don't care too much about where the waypoints are, another way to make an 8 figure would be to use the offboard mode, I would say.

MehdiRifi, 2022-02-21:
Fixed wings vehicles can't hover on a position like copter, so only 4 choices are possible:

- loiter around the given waypoint
- continue to the next waypoint 
- land at the given waypoint
- return to launch

when planning a mission pay attention to the radius of the waypoints if it is big enough that circles will cut each other you will get the something similar as you're last picture.

decrease the waypoint radius, no idea where to find it in qgroundcontrol but in mission planner you will find it as follow :
[image]

the bigger the distance the accurate the behavior you will have based on my experience.

julianoes, 2022-03-02:
I think you need to try to make the plan bigger, or reduce the radius that the plane can fly. I assume you are using ArduPilot, and not PX4?

Essentially, if you can make it work with MissionPlanner, then you should be able to set a mission with the exact same mission items using the `MissionRaw` plugin.

julianoes, 2022-04-05:
There was no more follow up, so I'm closing this. Just comment again if this is still unresolved.

closed 2022-04-05 by julianoes