mavlink/MAVSDK-Python #445: Not receiving output for certain function calls.

matthew-ay, 2022-02-22:
Hello, 

I'm running into an issue when running the telemetry.py file found in the examples directory. Specifically, the function call 'drone.telemetry.position()' does not produce an output to the terminal window, whereas the other functions in the file do produce an output. 

I installed Mavsdk-python using pip3. I manually run the mavsdk-server in a separate terminal window with this command: 

```
./mavsdk_server -p 50051 serial:///dev/tty.usbmodem01
```

And then run the telemetry.py script with the following line changed: 

```
await drone.connect(system_address="serial:///dev/tty.usbmodem01")
```

After starting the script, I receive the following output from mavsdk_server: 

```
[05:11:03|Info ] MAVSDK version: v0.50.0 (mavsdk_impl.cpp:28)
[05:11:03|Info ] Waiting to discover system on serial:///dev/tty.usbmodem01... (connection_initiator.h:20)
[05:11:03|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:484)
[05:11:03|Debug] Component Autopilot (1) added. (system_impl.cpp:355)
[05:11:03|Debug] Discovered 1 component(s) (system_impl.cpp:523)
[05:11:03|Info ] System discovered (connection_initiator.h:63)
[05:11:03|Info ] Server started (grpc_server.cpp:52)
[05:11:03|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:53)
[05:11:11|Info ] heartbeats timed out (system_impl.cpp:257)
[05:11:13|Debug] Discovered 1 component(s) (system_impl.cpp:523)
[05:11:17|Info ] heartbeats timed out (system_impl.cpp:257)
[05:11:19|Debug] Discovered 1 component(s) (system_impl.cpp:523)
[05:11:20|Warn ] sending again after 0.500246 s, retries to do: 3  (520). (mavlink_commands.cpp:275)
[05:11:20|Warn ] sending again after 1.01174 s, retries to do: 2  (520). (mavlink_commands.cpp:275)
[05:11:23|Info ] heartbeats timed out (system_impl.cpp:257)
[05:11:24|Debug] Discovered 1 component(s) (system_impl.cpp:523)
```

And partial output from the terminal window running telemetry.py: 

```
In air: False
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -0.009999999776482582
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -0.009999999776482582
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -0.009999999776482582
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -0.009999999776482582
```

My question: is there a reason why this section of telemetry.py doesn't produce output to the terminal? 

```
async def print_position(drone):
    async for position in drone.telemetry.position():
        print(position)
```

I ran into the same issue with drone.telemetry.heading(). For both cases, is the issue due to the fact that the GPS module can't reach a satellite? I have completed the calibration for the GPS module using QGroundControl. 

System info: 
Running on macOS Catalina (ver. 10.15.7). 
Python 3.9.10. 
PX4 FMU V2, running PX4 Pro Stable Release v1.12. 
Attached is a 3D Robotics GPS Module for Pixhawk.

MehdiRifi, 2022-02-22:
I think there is something not working correctly in  the device connection , the server terminal keeps printing heartbeat time out , and even for gps fix it indicates no gps .
Could u connect to ur device using qgroundcontrol or mission planner and see if you will get gps infos ??

matthew-ay, 2022-02-22:
@MehdiRifi Thank you for the response. I plugged the Pixhawk into a seperate machine running QGC, ensured that the GPS module is plugged in, and then went to: Select Tool menu -> Analyze Tools -> MAVLink Inspector -> GPS_RAW_INT. 

The value for Count and time_usec are increasing, while every other value remains the same. If I unplug the GPS module while inspecting the GPS data, Count and time_use stop. If I plug the GPS module back in, they continue to increase. The value of "satellites_visible" remains 0 in this case as well. 

I also looked at the status of the GPS under the MAVLink Console. "sat info" is being output as "disabled", so it is unclear whether that is also part of the issue. 

[image]

[image]

MehdiRifi, 2022-02-24:
Sorry for the late response, hmm seems weird to me, did u try:
test another gps module to ensure gps module is working fine !? 
Are you indoor ??

matthew-ay, 2022-02-25:
@MehdiRifi No worries.

I don't have another GPS module available, so I couldn't give that one a try. 

I had tested outside prior but was getting the same messages. However, I tried again, but this time much further away from the building. Fortunately, that proved to be the trick and fixed the issues I was having. Thank you for your help with this issue. 

If it is not too uncourteous, I have one more quick question that I was not able to solve by referencing the [MAVSDK-Python documentation](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/index.html). 

If I want to print the number of satellites using the Telemetry plugin, I use the following code: 

```
async def print_num_satellites(drone):
    async for gps_info in drone.telemetry.gps_info():
        print(f"GPS info: {gps_info.num_satellites}")
```

The above code now works, so long as I'm outside the building. However, if I want to print the latitude of the Pixhawk, I assumed I would follow the same name convention as the "GpsInfo" class. As such, I use the following code: 

```
async def print_latitude(drone):
        async for gps_global_origin in drone.telemetry.gps_global_origin():
            print(f"Latitude: {gps_global_origin.latitude_deg}")
```

I get the following error, though: 

```
Task exception was never retrieved
future: <Task finished name='Task-3' coro=<print_latitude() done, defined at /Users/user/Documents/example_heading.py:20> exception=AttributeError("'Telemetry' object has no attribute 'gps_global_origin'")>
Traceback (most recent call last):
  File "/Users/user/Documents/example_heading.py", line 21, in print_latitude
    async for gps_global_origin in drone.telemetry.gps_global_origin():
AttributeError: 'Telemetry' object has no attribute 'gps_global_origin'
```

Same happens if I attempt to access the Telemetry object with "drone.telemetry.GpsGlobalOrigin()". Both fail when I'm outside. Could you provide insight on the mistake I am making here?

MehdiRifi, 2022-02-25:
Latitude, Longitude and Altitude are available at The Position object , for getting latitude you should use:
```
async def print_position(drone):
    async for position in drone.telemetry.position():
        print(position.latitude_deg)
```
here is what available at the position class:
[image]

PS: the problem was that there is no attribute called 'gps_global_origin' in the telemetry object, here is the full documentation of the telemetry class [Telemetry Definition](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html)

matthew-ay, 2022-02-25:
@MehdiRifi Awesome, thank you. 

In the documentation, I also found this command in the Telemetry class that addresses the GpsGlobalOrigin() attribute. However, I'll be using Position() instead, so I won't be using the attribute.  

```
get_gps_global_origin()
```

Thank you again MehdiRifi for all of your help with this issue. 

Stay safe, 
Matt

closed 2022-02-25 by matthew-ay

MehdiRifi, 2022-03-01:
@matthew-ay no problem, my pleasure