mavlink/MAVSDK-Python #447: Is manual landing possible

ltrogers98, 2022-02-23:
Hello,

For some reason, the land command is not working. It just keeps saying command denied. This is the same for the takeoff. However, I can manually takeoff by throttling up. The drone recognizes this as taking off. Is it possible to manually land and have the drone recognize this as well. I am using custom code with a joystick hooked to my PC. Currently, I am trying to throttle down, but nothing ever happens.

JonasVautherin, 2022-02-23:
If commands are always denied, it means that you have a problem somewhere that needs to be fixed. Trying to bypass it with manual control will not fix it, right?

ltrogers98, 2022-02-24:
I can throttle up manually and the drone will register as it has taken off. I'm assuming the same is possible for landing. I'm wondering if anyone has any success with this. 

As for the command denied, it can arm, but the drone.action.land() and drone.action.takeoff() commands both throw command denied. This occurs in both Qgroundcontrol and custom code. I'm assuming there is a hardware issue? My understanding is that these commands should work no issues with custom code.

JonasVautherin, 2022-02-24:
> My understanding is that these commands should work no issues with custom code.

If they don't work with QGC, it means that the autopilot denies them for some reason. That's something to check on the autopilot side, I would say.

ltrogers98, 2022-02-24:
> > My understanding is that these commands should work no issues with custom code.
> 
> If they don't work with QGC, it means that the autopilot denies them for some reason. That's something to check on the autopilot side, I would say.

Thank you. Do you know a good link to explore this, or someone to reach out to?