mavlink/MAVSDK-Python #448: Datalink RC failsafe issue

m-sienk, 2022-03-01:
I have a scenario where I have built my own basic GCS (a web-app) and want to control a drone over the LTE network, currently testing using Gazebo sim and a raspi+pixhawk.

[Web-app] -> (sends commands via MQTT) -> [Drone-client, running mavsdk-python] 

When using Gazebo sim, in order to issue commands from my drone-client to the simulated aircraft I need QGroundControl running at the same time in order to create a datalink evidenced by the warning: `WARN  [commander] Failsafe enabled: no datalink`
To get around this, meaning I don't require QGroundControl running, I set `NAV_RCL_ACT` to 0 so no action is taken when RC link loss occurs.

This however creates issues with the flight mode, take the following example.

```
import asyncio
from mavsdk import System

aircraft = System()

async def flight_mode_updates():
    async for flight_mode in aircraft.telemetry.flight_mode():
        print(flight_mode)

async def main():
    # Aircraft connect
    await aircraft.connect(system_address="udp://:14540")

    # Get fmode status
    asyncio.create_task(flight_mode_updates())

    # Set NAV_RCL_ACT to 0
    await aircraft.param.set_param_int("NAV_RCL_ACT", 0)
    print(f"NAV_RCL_ACT = {await aircraft.param.get_param_int('NAV_RCL_ACT')}")

    await asyncio.sleep(5)

    # Aircraft takeoff
    await aircraft.action.arm()
    await aircraft.action.takeoff()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    try:
        loop.create_task(main())
        loop.run_forever()
    finally:
        loop.stop()
        loop.close()
```

If I run the following example without also running QGroundControl at the same time, the aircraft will arm and takeoff but the flight mode will never reach 'HOLD' it will remain on 'TAKEOFF'.
If however QGroundControl is also running it performs as expected.

Note I used the docker image to test: `docker run --rm -it -p 8554:8554 jonasvautherin/px4-gazebo-headless:1.12.1 -v typhoon_h480`

julianoes, 2022-03-02:
> in order to create a datalink evidenced by the warning

If the MAVSDK script is running for the entire duration it should always be sending heartbeats to the autopilot and you should have a valid datalink. I would look into what is going on there. You can try to debug what mavlink connections you have in the PX4 shell (`pxh>`) using `mavlink status`.

> will never reach 'HOLD' it will remain on 'TAKEOFF'.

Even if you leave the mavsdk script open for a while longer with some sleeps?

JonasVautherin, 2022-03-03:
I can actually reproduce :thinking:. Without QGC running, trying to takeoff and arm:

```INFO  [commander] Armed by external command
INFO  [navigator] Using minimum takeoff altitude: 2.50 m
INFO  [commander] Takeoff detected
WARN  [commander] Failsafe enabled: no datalink
INFO  [commander] Failsafe mode activated
INFO  [tone_alarm] battery warning (fast)
INFO  [navigator] RTL HOME activated
INFO  [navigator] RTL: landing at home position.
INFO  [navigator] RTL: climb to 489 m (1 m above destination)
INFO  [commander] Failsafe mode deactivated
INFO  [navigator] RTL: return at 489 m (1 m above destination)
INFO  [navigator] RTL: land at destination
INFO  [commander] Landing detected
INFO  [commander] Disarmed by landing```

So it does not even takeoff.

With QGC running:

```INFO  [commander] Armed by external command
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-03-03/10_25_15.ulg
INFO  [logger] Opened full log file: ./log/2022-03-03/10_25_15.ulg
INFO  [navigator] Using minimum takeoff altitude: 2.50 m
INFO  [commander] Takeoff detected```

julianoes, 2022-03-03:
Thanks for reproducing! Does PX4 somehow not "accept" the heartbeats that MAVSDK is sending? That would not be good...

julianoes, 2022-03-03:
Ok, I can reproduce it with MAVSDK-Python, however, it works with MAVSDK.

The reason for PX4 to complain about "no datalink" is based on the heartbeats it receives. Specifically, it looks for a heartbeat with `MAV_TYPE_GCS` (https://github.com/PX4/PX4-Autopilot/blob/490a0c473b9787e1133c95cfa17804dae87592ca/src/modules/mavlink/mavlink_receiver.cpp#L2116_https://github.com/PX4/PX4-Autopilot/blob/490a0c473b9787e1133c95cfa17804dae87592ca/src/modules/mavlink/mavlink_receiver.cpp#L2116).

If MAVSDK is configured as a ground station, then it should send `MAV_TYPE_GCS`, however, for some reason this might not be the case for `masvdk_server` as part of Python. I'm checking that next.

julianoes, 2022-03-04:
Ok, I found the issue. This is a regression since https://github.com/mavlink/MAVSDK/pull/1577.

It will be fixed with https://github.com/mavlink/MAVSDK/pull/1713.

closed 2022-03-06 by julianoes

mengchaoheng, 2022-10-23:
The current flight failsafe strategy is very unreasonable, I can't take off many planes at the same time, because I can't ensure multiple data links in the simulation @julianoes

julianoes, 2022-10-24:
PX4 failsafes have improved considerably in the PX4 main branch, have a look at:

https://github.com/PX4/PX4-Autopilot/pull/20172