mavlink/MAVSDK-Python #449: ModuleNotFoundError: No module named 'mavsdk' when install in raspbian pi 4

huytd-pi, 2022-03-14:
I install mavsdk on my Raspberry pi 4. I installed on mavsdk with python3, my python3 version is 3.7.3 and has been set as compile default, but when compiling No module named 'mavsdk'. Has anyone had the same problem and how did you fix it? thanks

**pi@droneraspi:~/Desktop/huytd/MAVSDK-Python/examples $ python2**
Python 2.7.16 (default, Oct 10 2019, 22:02:15) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mavsdk
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named mavsdk
>>> exit()
**pi@droneraspi:~/Desktop/huytd/MAVSDK-Python/examples $ python3**
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mavsdk
>>> exit()
**pi@droneraspi:~/Desktop/huytd/MAVSDK-Python/examples $ sudo python telemetry.py** 
Traceback (most recent call last):
  File "telemetry.py", line 5, in <module>
    from mavsdk import System
ModuleNotFoundError: No module named 'mavsdk'
**pi@droneraspi:~/Desktop/huytd/MAVSDK-Python/examples $ sudo python3 telemetry.py** 
Traceback (most recent call last):
  File "telemetry.py", line 5, in <module>
    from mavsdk import System
ModuleNotFoundError: No module named 'mavsdk'

MehdiRifi, 2022-03-14:
Either :
- re-install **_mavsdk_** with _sudo_ 
- or run the script file without sudo

If u installed a package without sudo and run a program that import the same package with sudo, python will try to look for the package in a different place so it will not find it .

julianoes, 2022-03-14:
Why do you run it with `sudo` in the first place?

huytd-pi, 2022-03-15:
> Why do you run it with `sudo` in the first place?
I use uart on ras pi connect with pixhawk 4

huytd-pi, 2022-03-15:
> Either :
> 
> * re-install **_mavsdk_** with _sudo_
> * or run the script file without sudo
> 
> If u installed a package without sudo and run a program that import the same package with sudo, python will try to look for the package in a different place so it will not find it .

thank you! i fixed it

julianoes, 2022-03-15:
>  I use uart on ras pi connect with pixhawk 4

You should correctly set up the permissions for the user then:
https://askubuntu.com/a/58122