mavlink/MAVSDK-Python #450: Parallel threaded execution, error: command denied

lazaro-fh, 2022-03-14:
We are trying to launch two drones by means of two threads running the same module in parallel. 
In this module we perform the connection, arming and takeoff of the drone, but when the second thread tries to arm its drone we get the following error: command denied. This is the creation code of each thread:
```
            mission.name = "Drone"+str(i) 
            self.threads.append(mission)
            self.printLog(message="Starting missions: ", typeMessage="info")
            for mission in self.threads:
                mission.start()
```
mission.start() runs another module called "takeoff.py" with the following code:
```
 # Connect this Drone to the MavSDK-Server
    portBase  = 50050
    portBase2 = 14550
    port = portBase2 + self.drone_id #We use different ports for each drone.

    self.printLog(message="Drone"+str(self.drone_id)+" connecting to port "+str(port)+"...", typeMessage="info")

    self.drone = System()

    await self.drone.connect(system_address="udp:#:"+str(port))
    self.printLog(message="-- Arming "+str(self.drone_id), typeMessage="info")
    with self.lock:
      try:
        await self.drone.action.arm()
      except ActionError as error:
        self.printLog(message="-- Arm failed with error code "+str(error), typeMessage="info")
        self.printLog(message="-- error._result.result "+str(self.drone_id), typeMessage="info")
        await self.drone.action.disarm()
        return
    await asyncio.sleep(1)
```
however, when we run everything on a single module we have no problem. An example would be if mission.start() executes takeoff(), which is an asynchronous function of the same module where the thread was created.
We have a PX4 and a MavSDK for each drone, which are recompiled in each execution. 
Does anyone know why this happens and how to fix it? Thank you very much

julianoes, 2022-03-15:
I think you need to make sure that actually two mavsdk_server processes are started in the background.

If you share your full example, I can try to run it and see what is going on.

julianoes, 2022-04-05:
I assume you have resolved this given there was no response and I'm closing the issue. If not, just comment again here, thanks.

closed 2022-04-05 by julianoes