mavlink/MAVSDK-Python #451: Error cannot run examples

huytd-pi, 2022-03-15:
I make run the examples of python mavsdk
some examples like takeoff_and_land.py, telemetry.py, telemtry_flight_mode.py ... are examples that can be run normally
But examples like offboard_*.py, maual_control.py, follow_me_example.py .. can't execute, drone in simulator only does arm, takeoff, land, in terminal only shows printout messages, do not execute set_attitude, set_velocity_ned ...

```
huytd@huytd-HP:~/Project/Drone/MAVSDK/MAVSDK-Python/examples$ python3 takeoff_and_land.py 
Waiting for drone to connect...
Drone discovered!
Waiting for drone to have a global position estimate...
Global position estimate ok
-- Arming
-- Taking off
-- Landing
huytd@huytd-HP:~/Project/Drone/MAVSDK/MAVSDK-Python/examples$ python3 offboard_attitude.py 
Waiting for drone to connect...
Drone discovered!
-- Arming
-- Setting initial setpoint
-- Starting offboard
-- Go up at 70% thrust
-- Roll 30 at 60% thrust
-- Roll -30 at 60% thrust
-- Hover at 60% thrust
-- Stopping offboard
```

I don't know what's wrong with it, need your answers!

julianoes, 2022-03-15:
Could you share the output on the PX4 side please?

huytd-pi, 2022-03-16:
> Could you share the output on the PX4 side please?

```
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/huytd/Project/Drone/PX4...e/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/huytd/Project/Drone/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/huytd/Project/Drone/PX4-Autopilot
build_path: /home/huytd/Project/Drone/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/home/huytd/Project/Drone/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/huytd/Project/Drone/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /opt/ros/melodic/lib:/home/huytd/Project/Drone/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /home/huytd/Project/Drone/PX4-Autopilot/Tools/sitl_gazebo/models/iris/iris.sdf
SITL COMMAND: "/home/huytd/Project/Drone/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/huytd/Project/Drone/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/huytd/Project/Drone/PX4-Autopilot"/test_data
Creating symlink /home/huytd/Project/Drone/PX4-Autopilot/build/px4_sitl_default/etc -> /home/huytd/Project/Drone/PX4-Au

[0______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
INFO  [parameters] BSON document size 806 bytes, decoded 806 bytes (INT32:22, FLOAT:17)
[param] Loaded: eeprom/parameters_10016
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [navigator] Mission #3 loaded, 7 WPs
Gazebo multi-robot simulator, version 9.19.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.105
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [init] setting PWM_AUX_OUT none
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-03-16/02_12_47.ulg	
INFO  [logger] Opened full log file: ./log/2022-03-16/02_12_47.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 127.0.0.1
INFO  [tone_alarm] home set
WARN  [mc_pos_control] invalid setpoints
INFO  [tone_alarm] notify positive

```

julianoes, 2022-03-16:
Ok, that looks ok. I'll try to reproduce that tomorrow.

huytd-pi, 2022-03-17:
> Ok, that looks ok. I'll try to reproduce that tomorrow.

I'm waiting for your reply, ,thanks!

JonasVautherin, 2022-03-17:
> WARN  [mc_pos_control] invalid setpoints

What's that?

julianoes, 2022-03-17:
@JonasVautherin it's noise, you find it everywhere. It's race conditions in the PX4 pubsub architecture, so some setpoints are sometimes NAN initially. It just gets ignored.

julianoes, 2022-03-17:
@JonasVautherin you might be right that this is related to https://github.com/mavlink/MAVSDK/pull/1713. Let me try to update MAVSDK-Python with the latest and greatest.

julianoes, 2022-03-18:
I made a release. @huytd-pi can you check with mavsdk 1.0.0? Some examples might still not work against latest PX4 as it had some failsafe and preflight check changes that we need to work through.

closed 2022-08-08 by huytd-pi