mavlink/MAVSDK-Python #454: When set GimbalMode.YAW_FOLLOW , there is crash!!!

msasen, 2022-03-17:
[image]

I want to look at the determined roi region with fixed camera but I am getting this error.

julianoes, 2022-03-18:
That's not nice :astonished:, let me check that.

julianoes, 2022-03-18:
So I just ran the gimbal example against PX4 SITL with the typhoon_h480 model and it all worked, this is with latest MAVSDK-Python 1.0.0 version. Can you double check that?

And if it fails, could you share the exact steps or script that I should try to reproduce it? Thank you.

julianoes, 2022-03-23:
Ok, and you're trying this against what, SITL? And if so, which model?

Sitin, 2022-06-21:
I am using MAVSDK-Puthon 1.3.0 and ArduPilot SITL with `ArduCopter/+` vehicle and have the same error. Everything was configured according to the [official manual](https://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html).

Sitin, 2022-06-21:
Thanks, @julianoes. What do you suggest to use to test gimbal controls?

julianoes, 2022-06-22:
So I would expect that the gimbal v1 protocol works for ArduPilot, however, the mode command might not be implemented, or implemented in the same way. If you figure out how/if ArduPilot allows you to change the mode, then we can implement that in MAVSDK.

Note that gimbal is a bit of a sad topic, as it was - a long time ago - decided on the MAVLink call that there should be a new protocol, which I then developed but which is only implemented and accepted very slowly, if at all.