mavlink/MAVSDK-Python #458: MAVSDK Python -- Flying an imported Q Ground Control Mission

StavrosKontzias29, 2022-03-27:
Hello,

Working with MAVSDK for my senior capstone project. Tested multiple functions with MAVSDK such as arming the drone and the  Take Off and Land example given in the MAVSDK Python GitHub Page.  Currently, I am able to import a Q Ground Control Mission, but I am having issues to actually begin that flight mission. Figured I should just be able to run the "start_mission()" function but no luck and I am stuck. Any help would be appreciated!

Thank you

julianoes, 2022-03-28:
Hi there,

What error do you see in Python when you try to start the mission? And if there error doesn't tell you anything, then you can try to get more output by following these steps: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

StavrosKontzias29, 2022-03-28:
So I originally tried to use the the line  "  await drone.mission.start_mission(mission_import_data) " assuming that would run the mission. I do know this is incorrect, the error I got from python was " await drone.mission.start_mission(mission_import_data) NameError: name 'mission_import_data' is not defined. I've also attached the code I am using. We are using a raspberry pi 4 as an onboard computer to run the code. Like I mentioned before we were able to run the takeoff and Land example on our drone  without any issues so am assuming I am missing something needed to actually begin the flight of the QGC mission in my code.
 Below is the code we have been trying to get to work
Thank you again!

import asyncio
from mavsdk import System

drone = System()

# Connect to Flight Controller on ACM0
async def connect():
    print("--Looking for the drone now")
    await drone.connect(system_address="serial:///dev/ttyACM0")

# Verify Connection to Flight Controller
async def check_drone_connection():
    await connect()
    print("---Waiting for drone.")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"--Drone discovered")
            break

# Check GPS Connection
async def check_gps():
    await connect()
    print("--Waiting for drone to have global position")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok:
            print(f"--Global position estimate ok")
            break

# Arm Drone
async def arm_drone():
    print("-- Arming")
    await drone.action.arm()

# Import QGC flight mission
async def import_flight_mission():
    print("-- Uploading Mission")
    mission_import_data = await drone.mission_raw.import_qgroundcontrol_mission("flight_mission.plan")
    print(f"{len(mission_import_data.mission_items)} mission items imported")
    await drone.mission_raw.upload_mission(mission_import_data.mission_items)
    print("-- Mission uploaded")

# Run imported mission
async def run_imported_mission():
    print("-- Starting Mission")
    await drone.mission.start_mission()

# Run Drone
async def run_drone():
    await connect()
    await check_drone_connection()
    await import_flight_mission()
    await arm_drone()
    await run_imported_mission()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run_drone())

julianoes, 2022-03-29:
Ok, so what error do you see with this?

And have you tried to get more debugging output as described in: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

StavrosKontzias29, 2022-03-29:
Yes I did try the steps on the link you attached and we don’t have any connection issues. I’m attaching a picture of the error output I meant to put that in my last post. Sorry that the image is a little blurry

[image]

julianoes, 2022-03-29:
You know you can take screenshots directly ;) Can you run mavsdk_server separate and share the output of it? That would probably help.

StavrosKontzias29, 2022-03-29:
Haha ye I am well aware I was in a hurry and all I had was the photo on my phone. But here is the output for mavsdk_server

[image]
r

StavrosKontzias29, 2022-03-30:
So we ran our code and mavsdk_server separately while it was running and this is the output on the server for the first image. Second image is just the full output from mavsdk_server

[image]
[image]
r

julianoes, 2022-03-30:
Ok please also check if this is still the same with the latest version.

This should update it:
```
pip3 install --user --upgrade mavsdk
```

StavrosKontzias29, 2022-04-01:
I was able to update mavsdk. Below is what mavsdk_server was showing. Key one I noticed was "Mission rejected: takeoff altitude too low".
[image]

julianoes, 2022-04-03:
Ok, what is the takeoff altitude that you're trying to send? Or is maybe the frame incorrect?

julianoes, 2022-04-05:
@StavrosKontzias29 can you share the QGC plan that you're using? That would allow me to reproduce it.

StavrosKontzias29, 2022-04-10:
Hello, Sorry for the late response. We had the takeoff altitude too low once we changed it we were able to successfully fly the mission!!

closed 2022-04-10 by julianoes