mavlink/MAVSDK-Python #460: Serial connection error using Raspi

m-sienk, 2022-03-31:
Currently I'm working with a Raspberry Pi 4B and a Pixhawk 4, the Pi and Pixhawk are communicating over a serial connection which connects a USB port on the Pi to the telem2 port on the Pixhawk.

Here are some relevant settings:
```
SER_TEL1_BAUD = 57600 8N1
SER_TEL2_BAUD = 921600 8N1
MAV_0_CONFIG = TELEM1
MAV_1_CONFIG = TELEM1
MAV_2_CONFIG = TELEM2
```

The serial connection works, however whenever I first run the solution on the Raspi via `python3 main.py` it results in the following error.

```
__anext__ grpc exception
Traceback (most recent call last):
  File "/home/pi/connection-test/venv/lib/python3.9/site-packages/aiogrpc/utils.py", line 145, in cb
    fut.result()
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/connection-test/venv/lib/python3.9/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/pi/connection-test/venv/lib/python3.9/site-packages/grpc/_channel.py", line 426, in __next__
    return self._next()
  File "/home/pi/connection-test/venv/lib/python3.9/site-packages/grpc/_channel.py", line 826, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Connection reset by peer"
	debug_error_string = "{"created":"@1648686218.868823056","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":903,"grpc_message":"Connection reset by peer","grpc_status":14}"
```

If I then stop the solution `ctrl+c` then run it again `python3 main.py` it works flawlessly. Every subsequent run the solution works as expected it is only the initial run that causes this error.

If I then turn off the Raspi and take the same course of action the same pattern follows.

**main.py**
```
import asyncio
from mavsdk import System

system = System()

async def check_connection_state():
    print("Attempting to connect to flight controller")
    async for connection_state in system.core.connection_state():
        if connection_state.is_connected:
            print("Success, connected to flight controller")
        else:
            print("Disconnected from flight controller")

async def get_armed_status():
    async for armed in system.telemetry.armed():
        print(armed)

async def get_angular_velocity_body_status():
    async for attitude_angular_velocity_body in system.telemetry.attitude_angular_velocity_body():
        print(attitude_angular_velocity_body)

async def get_attitude_euler_status():
    async for attitude in system.telemetry.attitude_euler():
        print(attitude)

async def get_battery_status():
    async for battery in system.telemetry.battery():
        print(battery)

async def get_flight_mode_status():
    async for flight_mode in system.telemetry.flight_mode():
        print(flight_mode)

async def get_gps_info_status():
    async for gps_info in system.telemetry.gps_info():
        print(gps_info)

async def get_health_status():
    async for health in system.telemetry.health():
        print(health)

async def get_health_all_ok_status():
    async for health_all_ok in system.telemetry.health_all_ok():
        print(health_all_ok)

async def get_home_status():
    async for home in system.telemetry.home():
        print(home)

async def get_in_air_status():
    async for in_air in system.telemetry.in_air():
        print(in_air)

async def get_landed_state_status():
    async for landed_state in system.telemetry.landed_state():
        print(landed_state)

async def get_position_status():
    async for position in system.telemetry.position():
        print(position)

async def get_velocity_ned_status():
    async for velocity_ned in system.telemetry.velocity_ned():
        print(velocity_ned)

async def main():
    await system.connect("serial:///dev/ttyUSB0:921600")
    asyncio.create_task(check_connection_state())
    asyncio.create_task(get_armed_status())
    asyncio.create_task(get_angular_velocity_body_status())
    asyncio.create_task(get_attitude_euler_status())
    asyncio.create_task(get_battery_status())
    asyncio.create_task(get_flight_mode_status())
    asyncio.create_task(get_gps_info_status())
    asyncio.create_task(get_health_status())
    asyncio.create_task(get_health_all_ok_status())
    asyncio.create_task(get_home_status())
    asyncio.create_task(get_in_air_status())
    asyncio.create_task(get_position_status())
    asyncio.create_task(get_velocity_ned_status())

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    try:
        loop.create_task(main())
        loop.run_forever()
    finally:
        loop.stop()
        loop.close()
```
Could someone provide some insight as to why this is occurring?

JonasVautherin, 2022-03-31:
Seems like mavsdk_server is not running. Can you try running it manually?

julianoes, 2022-03-31:
I've now seen this several times. I wonder what causes this. Let me dig out a RPi and try to reproduce it.

In the meantime, could you test with latest MAVSDK-Python 1.0.0 and check if it also happens when you start mavsdk_server separately like explained here:
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

julianoes, 2022-03-31:
Reboot, and gdb, and here we go:

```
Thread 13 "mavsdk_server" received signal SIGSEGV, Segmentation fault.
[Switching to LWP 1207]
0x0000007ff7aead3c in pthread_mutex_lock ()
(gdb) bt
#0  0x0000007ff7aead3c in pthread_mutex_lock ()
#1  0x0000007ff714a634 in mavsdk::TelemetryImpl::subscribe_battery(std::function<void (mavsdk::Telemetry::Battery)>&) ()
#2  0x0000007ff700fda8 in mavsdk::mavsdk_server::TelemetryServiceImpl<mavsdk::Telemetry, mavsdk::mavsdk_server::LazyPlugin<mavsdk::Telemetry> >::SubscribeBattery(grpc::ServerContext*, mavsdk::rpc::telemetry::SubscribeBatteryRequest const*, grpc::ServerWriter<mavsdk::rpc::telemetry::BatteryResponse>*) ()
#3  0x0000007ff73a29ec in std::_Function_handler<grpc::Status (mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, mavsdk::rpc::telemetry::SubscribeBatteryRequest const*, grpc::ServerWriter<mavsdk::rpc::telemetry::BatteryResponse>*), mavsdk::rpc::telemetry::TelemetryService::Service::ServerWriter()::{lambda(mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, mavsdk::rpc::telemetry::SubscribeBatteryRequest const*, grpc::ServerWriter<mavsdk::rpc::telemetry::BatteryResponse>*)#15}>::_M_invoke(std::_Any_data const&, mavsdk::rpc::telemetry::TelemetryService::Service*&&, grpc::ServerContext*&&, mavsdk::rpc::telemetry::SubscribeBatteryRequest const*&&, grpc::ServerWriter<mavsdk::rpc::telemetry::BatteryResponse>*&&) ()
#4  0x0000007ff7444bb0 in grpc::internal::ServerStreamingHandler<mavsdk::rpc::telemetry::TelemetryService::Service, mavsdk::rpc::telemetry::SubscribeBatteryRequest, mavsdk::rpc::telemetry::BatteryResponse>::RunHandler(grpc::internal::MethodHandler::HandlerParameter const&) ()
#5  0x0000007ff74ec6d8 in grpc::Server::SyncRequest::ContinueRunAfterInterception() ()
#6  0x0000007ff74f3d3c in grpc::ThreadManager::MainWorkLoop() ()
#7  0x0000007ff74f3f08 in grpc::ThreadManager::WorkerThread::Run() ()
#8  0x0000007ff784bd30 in grpc_core::(anonymous namespace)::ThreadInternalsPosix::ThreadInternalsPosix(char const*, void (*)(void*), void*, bool*, grpc_core::Thread::Options const&)::{lambda(void*)#1}::_FUN(void*) ()
#9  0x0000007ff7aea444 in start ()
#10 0x0000007ff7aea444 in start ()
```

Three things are odd.

1. Why are there debugging symbols? 😲 This should be a release build. 
2. Why does it only happen on the first try? Is it just timing? Somewhat odd.
3. Why does it crash on locking? Usually it means we already own the lock in the same thread. Let's have a look.

julianoes, 2022-03-31:
I don't see it. I will build a debug build on the RPi4 overnight, hopefully that will show more.

julianoes, 2022-03-31:
But it looks like this might be an unrelated error as brought up here as well https://github.com/grpc/grpc/issues/29098.

julianoes, 2022-03-31:
Ok, when running the debug build in gdb I was able to reproduce it and get a backtrace. Presumably running it in gdb made it a bit slower and changed the timing just enough to make it happen, or I was just lucky this time.

```
(gdb) bt
#0  0x0000007ff47d6c30 in __GI___pthread_mutex_lock (mutex=0x10000082a) at pthread_mutex_lock.c:67
#1  0x0000007ff6a49938 in __gthread_mutex_lock (__mutex=0x10000082a) at /usr/include/aarch64-linux-gnu/c++/10/bits/gthr-default.h:749
#2  0x0000007ff6a49b7c in std::mutex::lock (this=0x10000082a) at /usr/include/c++/10/bits/std_mutex.h:100
#3  0x0000007ff6a6155c in std::lock_guard<std::mutex>::lock_guard (this=0x7fd1ff9e68, __m=...) at /usr/include/c++/10/bits/std_mutex.h:159
#4  0x0000007ff4d1c4d0 in mavsdk::TelemetryImpl::subscribe_flight_mode(std::function<void (mavsdk::Telemetry::FlightMode)>&) (this=0x100000002, callback=...) at /home/pi/src/MAVSDK/src/mavsdk/plugins/telemetry/telemetry_impl.cpp:2277
#5  0x0000007ff4d0a268 in mavsdk::Telemetry::subscribe_flight_mode(std::function<void (mavsdk::Telemetry::FlightMode)>) (this=0x7fc0005150, callback=...) at /home/pi/src/MAVSDK/src/mavsdk/plugins/telemetry/telemetry.cpp:205
#6  0x0000007ff6a78ad0 in mavsdk::mavsdk_server::TelemetryServiceImpl<mavsdk::Telemetry, mavsdk::mavsdk_server::LazyPlugin<mavsdk::Telemetry> >::SubscribeFlightMode (this=0x55555a05f8, writer=0x7fd1ffa238)
    at /home/pi/src/MAVSDK/src/mavsdk_server/src/plugins/telemetry/telemetry_service_impl.h:1926
#7  0x0000007ff6dfdf6c in operator() (__closure=0x55555886c8, service=0x55555a05f8, ctx=0x7fc00048c0, req=0x7fc0003980, writer=0x7fd1ffa238) at /home/pi/src/MAVSDK/src/mavsdk_server/src/generated/telemetry/telemetry.grpc.pb.cc:1366
#8  0x0000007ff6e10c64 in std::__invoke_impl<grpc::Status, mavsdk::rpc::telemetry::TelemetryService::Service::Service()::<lambda(mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeFlightModeRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::FlightModeResponse>*)>&, mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeFlightModeRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::FlightModeResponse>*>(std::__invoke_other, struct {...} &) (__f=...) at /usr/include/c++/10/bits/invoke.h:60
#9  0x0000007ff6e0adec in std::__invoke_r<grpc::Status, mavsdk::rpc::telemetry::TelemetryService::Service::Service()::<lambda(mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeFlightModeRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::FlightModeResponse>*)>&, mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeFlightModeRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::FlightModeResponse>*>(struct {...} &) (__fn=...) at /usr/include/c++/10/bits/invoke.h:115
#10 0x0000007ff6e0541c in std::_Function_handler<grpc::Status(mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeFlightModeRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::FlightModeResponse>*), mavsdk::rpc::telemetry::TelemetryService::Service::Service()::<lambda(mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeFlightModeRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::FlightModeResponse>*)> >::_M_invoke(const std::_Any_data &, mavsdk::rpc::telemetry::TelemetryService::Service *&&, grpc::ServerContext *&&, const mavsdk::rpc::telemetry::SubscribeFlightModeRequest *&&, grpc::ServerWriter<mavsdk::rpc::telemetry::FlightModeResponse> *&&) (__functor=..., __args#0=@0x7fd1ffa190: 0x55555a05f8, __args#1=@0x7fd1ffa188: 0x7fc00048c0, __args#2=@0x7fd1ffa180: 0x7fc0003980, __args#3=@0x7fd1ffa178: 0x7fd1ffa238)
    at /usr/include/c++/10/bits/std_function.h:292
#11 0x0000007ff6eaf908 in std::function<grpc::Status (mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, mavsdk::rpc::telemetry::SubscribeFlightModeRequest const*, grpc::ServerWriter<mavsdk::rpc::telemetry::FlightModeResponse>*)>::operator()(mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, mavsdk::rpc::telemetry::SubscribeFlightModeRequest const*, grpc::ServerWriter<mavsdk::rpc::telemetry::FlightModeResponse>*) const (
    this=0x55555886c8, __args#0=0x55555a05f8, __args#1=0x7fc00048c0, __args#2=0x7fc0003980, __args#3=0x7fd1ffa238) at /usr/include/c++/10/bits/std_function.h:622
#12 0x0000007ff6e7ae88 in grpc::internal::ServerStreamingHandler<mavsdk::rpc::telemetry::TelemetryService::Service, mavsdk::rpc::telemetry::SubscribeFlightModeRequest, mavsdk::rpc::telemetry::FlightModeResponse>::RunHandler(grpc::internal::MethodHandler::HandlerParameter const&)::{lambda()#1}::operator()() const (this=0x55555886c0) at /home/pi/src/MAVSDK/build/third_party/install/include/grpcpp/impl/codegen/method_handler.h:207
#13 0x0000007ff6eaf940 in grpc::internal::CatchingFunctionHandler<grpc::internal::ServerStreamingHandler<mavsdk::rpc::telemetry::TelemetryService::Service, mavsdk::rpc::telemetry::SubscribeFlightModeRequest, mavsdk::rpc::telemetry::FlightModeResponse>::RunHandler(grpc::internal::MethodHandler::HandlerParameter const&)::{lambda()#1}>(grpc::internal::ServerStreamingHandler<mavsdk::rpc::telemetry::TelemetryService::Service, mavsdk::rpc::telemetry::SubscribeFlightModeRequest, mavsdk::rpc::telemetry::FlightModeResponse>::R [... cut ...]
#21 0x0000007ff6b1c734 in _FUN () at /home/pi/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/cpp/thread_manager/thread_manager.cc:35
#22 0x0000007ff74bc580 in operator() (__closure=0x0, v=0x7fbc004f90) at /home/pi/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/gprpp/thd_posix.cc:140
#23 0x0000007ff74bc5b8 in _FUN () at /home/pi/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/gprpp/thd_posix.cc:145
#24 0x0000007ff47d4628 in start_thread (arg=0x7fd1ffa9c0) at pthread_create.c:477
#25 0x0000007ff508301c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
```

julianoes, 2022-03-31:
So, this telemetry subscription (in another thread) seems to be correct:
```
Thread 15 (Thread 0x7fd17fa0c0 (LWP 1655) "grpcpp_sync_ser"):
#0  syscall () at ../sysdeps/unix/sysv/linux/aarch64/syscall.S:38
#1  0x0000007ff51eaab0 in std::__atomic_futex_unsigned_base::_M_futex_wait_until(unsigned int*, unsigned int, bool, std::chrono::duration<long, std::ratio<1l, 1l> >, std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /lib/aarch64-linux-gnu/libstdc++.so.6
#2  0x0000007ff6a6a8d4 in std::__atomic_futex_unsigned<2147483648u>::_M_load_and_test_until (this=0x7fcc006ee0, __assumed=0, __operand=1, __equal=true, __mo=std::memory_order_acquire, __has_timeout=false, __s=..., __ns=...) at /usr/include/c++/10/bits/atomic_futex.h:102
#3  0x0000007ff6a668e0 in std::__atomic_futex_unsigned<2147483648u>::_M_load_and_test (this=0x7fcc006ee0, __assumed=0, __operand=1, __equal=true, __mo=std::memory_order_acquire) at /usr/include/c++/10/bits/atomic_futex.h:121
#4  0x0000007ff6a4a03c in std::__atomic_futex_unsigned<2147483648u>::_M_load_when_equal (__mo=std::memory_order_acquire, __val=1, this=0x7fcc006ee0) at /usr/include/c++/10/bits/atomic_futex.h:162
#5  std::__future_base::_State_baseV2::wait (this=0x7fcc006ed0) at /usr/include/c++/10/future:337
#6  0x0000007ff6a918a8 in std::__basic_future<void>::wait (this=0x7fd17f8f00) at /usr/include/c++/10/future:703
#7  0x0000007ff6a78078 in mavsdk::mavsdk_server::TelemetryServiceImpl<mavsdk::Telemetry, mavsdk::mavsdk_server::LazyPlugin<mavsdk::Telemetry> >::SubscribeGpsInfo (this=0x55555a05f8, writer=0x7fd17f9238) at /home/pi/src/MAVSDK/src/mavsdk_server/src/plugins/telemetry/telemetry_service_impl.h:1823
#8  0x0000007ff6dfde58 in operator() (__closure=0x55555924a8, service=0x55555a05f8, ctx=0x7fb40048c0, req=0x7fb4003980, writer=0x7fd17f9238) at /home/pi/src/MAVSDK/src/mavsdk_server/src/generated/telemetry/telemetry.grpc.pb.cc:1336
#9  0x0000007ff6e10898 in std::__invoke_impl<grpc::Status, mavsdk::rpc::telemetry::TelemetryService::Service::Service()::<lambda(mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeGpsInfoRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::GpsInfoResponse>*)>&, mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeGpsInfoRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::GpsInfoResponse>*>(std::__invoke_other, struct {...} &) (__f=...) at /usr/include/c++/10/bits/invoke.h:60
#10 0x0000007ff6e0a864 in std::__invoke_r<grpc::Status, mavsdk::rpc::telemetry::TelemetryService::Service::Service()::<lambda(mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeGpsInfoRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::GpsInfoResponse>*)>&, mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeGpsInfoRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::GpsInfoResponse>*>(struct {...} &) (__fn=...) at /usr/include/c++/10/bits/invoke.h:115
#11 0x0000007ff6e04f90 in std::_Function_handler<grpc::Status(mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeGpsInfoRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::GpsInfoResponse>*), mavsdk::rpc::telemetry::TelemetryService::Service::Service()::<lambda(mavsdk::rpc::telemetry::TelemetryService::Service*, grpc::ServerContext*, const mavsdk::rpc::telemetry::SubscribeGpsInfoRequest*, grpc::ServerWriter<mavsdk::rpc::telemetry::GpsInfoResponse>*)> >::_M_invoke(const std::_Any_data &, mavsdk::rpc::telemetry::TelemetryService::Service *&&, grpc::ServerContext *&&, const mavsdk::rpc::telemetry::SubscribeGpsInfoRequest *&&, grpc::ServerWriter<mavsdk::rpc::telemetry::GpsInfoResponse> *&&) (__functor=..., __args#0=@0x7fd17f9190: 0x55555a05f8, __args#1=@0x7fd17f9188: 0x7fb40048c0, __args#2=@0x7fd17f9180: 0x7fb4003980, __args#3=@0x7fd17f9178: 0x7fd17f9238) at /usr/include/c++/10/bits/std_function.h:292
```
It seems to correctly wait for a futex here:
https://github.com/mavlink/MAVSDK/blob/cb2985dd2c8e9893a543d785d5ae56a08497d520/src/mavsdk_server/src/plugins/telemetry/telemetry_service_impl.h#L1823

Then the failing one has presumably acquired that mutex, so it might be the first one getting access to the telemetry plugin:

https://github.com/mavlink/MAVSDK/blob/cb2985dd2c8e9893a543d785d5ae56a08497d520/src/mavsdk_server/src/plugins/telemetry/telemetry_service_impl.h#L1926

Now something seems to be off with the lock that it's trying to acquire. The memory address doesn't look right:
```
#2  0x0000007ff6a49b7c in std::mutex::lock (this=0x10000082a) at /usr/include/c++/10/bits/std_mutex.h:100
```
And that's presumably why it crashes.

Now I wonder if that plugin is not fully instantiated and therefore the lock invalid, or if something else is overwriting that location in memory.

julianoes, 2022-04-01:
I'll close this as soon as the fixed Python version is released.

julianoes, 2022-04-01:
I hope it should be fixed, try 1.2.0, e.g. using `pip3 install --user --upgrade mavsdk`.

closed 2022-04-01 by julianoes

Jai-GAY, 2022-05-06:
it looks [like 1.3.1 is available](https://github.com/mavlink/MAVSDK/releases/tag/v1.3.1) why pip3 install --user --upgrade mavsdk still getting 1.2.0 instead, any idea?

julianoes, 2022-05-06:
Probably hasn't been updated, let me trigger that:
https://github.com/mavlink/MAVSDK-Python/pull/475

Jai-GAY, 2022-05-17:
I think is a bit messy here.

installation and show version is 1.3.0
pip3 show mavsdk version

WARNING: Package(s) not found: version
Name: mavsdk
**Version: 1.3.0**
Summary: Python wrapper for MAVSDK
Home-page: https://github.com/mavlink/MAVSDK-Python
Author: None
Author-email: None
License: UNKNOWN
Location: /home/pi/.local/lib/python3.9/site-packages
Requires: aiogrpc, grpcio, protobuf
Required-by: 

[Debug and test show](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues) 1.3.1.

~/.local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server serial:///dev/ttyAMA1:921600
[34m[02:02:17|Info ] [0m**MAVSDK version: v1.3.1** (mavsdk_impl.cpp:28)

julianoes, 2022-05-17:
Ok, so that looks correct.