mavlink/MAVSDK-Python #464: Flight mode does not change when calling system.action.goto_location

m-sienk, 2022-04-01:
When commanding a drone to go to a location using system.action.goto_location the drones flight mode does not change.

In the following example the drones flight mode is listed as 'HOLD' despite the drone moving to the set location.
Note: using mavsdk 1.2.0

```
import asyncio
from mavsdk import System

system = System()

async def get_flight_mode_status():
    async for flight_mode in system.telemetry.flight_mode():
        print(f"Flight mode: {flight_mode}")

async def main():
    await system.connect(system_address="udp://:14540")

    # print fmode
    asyncio.create_task(get_flight_mode_status())

    print("Waiting for drone to connect...")
    async for state in system.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in system.telemetry.health():
        if health.is_global_position_ok and health.is_home_position_ok:
            print("-- Global position state is good enough for flying.")
            break

    print("Fetching amsl altitude at home location....")
    async for terrain_info in system.telemetry.home():
        absolute_altitude = terrain_info.absolute_altitude_m
        break

    await system.action.arm()
    await system.action.takeoff()
    await asyncio.sleep(8)
    await system.action.goto_location(47.39852594514411, 8.545307673154218, absolute_altitude, 0)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    try:
        loop.create_task(main())
        loop.run_forever()
    finally:
        loop.stop()
        loop.close()
```

julianoes, 2022-04-03:
The way it is implemented in PX4 there is actually no goto flight mode. It just works in Hold mode (also called Loiter sometimes).

m-sienk, 2022-04-03:
Thanks for your response.

Does this not create an issue? When I send a 'go to' command to the drone I cannot then issue the 'hold' command to stop the drones action. The only way to cease this command is to issue another command such as land or perform a way-point mission etc.

julianoes, 2022-04-04:
That's a good point. I guess not :thinking:.