mavlink/MAVSDK-Python #468: `drone.offboard.stop()` gave us an error:

JinraeKim, 2022-04-07:
my coworkers and I tried to use offboard control using MAVSDK-Python.

The executed code is as follows (sorry for messy scripts).

```python3
#!/usr/bin/env python3

# Warning: Only try this in simulation!
#          The direct attitude interface is a low level interface to be used
#          with caution. On real vehicles the thrust values are likely not
#          adjusted properly and you need to close the loop using altitude.

from collections import namedtuple
import asyncio
import matplotlib.pyplot as plt

from mavsdk import System
from mavsdk.offboard import (Attitude, OffboardError)

AttitudeDatum = namedtuple("AttitudeDatum", [
    "roll",
    "m2",
    "m3",
    "m4",
])

ActuatorDatum = namedtuple("ActuatorDatum", [
    "m1",
    "m2",
    "m3",
    "m4",
])

class ActuatorDataBuffer:
    def __init__(self, name="buffer"):
        self.data = []
        self.name = name

    def append(self, m1, m2, m3, m4):
        datum = ActuatorDatum(m1, m2, m3, m4)
        self.data.append(datum)

    def __len__(self):
        return len(self.data)

    @property
    def m1(self):
        return [datum.m1 for datum in self.data]

    @property
    def m2(self):
        return [datum.m2 for datum in self.data]

    @property
    def m3(self):
        return [datum.m3 for datum in self.data]

    @property
    def m4(self):
        return [datum.m4 for datum in self.data]
        
    def plot(self):
        fig, ax = plt.subplots(4, 1)

        ax[0].plot(self.m1)
        ax[1].plot(self.m2)
        ax[2].plot(self.m3)
        ax[3].plot(self.m4)

        plt.savefig(self.name + ".png")
        plt.clf()
        plt.close()

async def run():
    """ Does Offboard control using attitude commands. """

    drone = System()
    data_buffer_1 = ActuatorDataBuffer(name="actuator_1")
    data_buffer_2 = ActuatorDataBuffer(name="actuator_2")

    await drone.connect(system_address="udp://:14540")
    # asyncio.ensure_future(print_actuator_output_status(drone, data_buffer_1, data_buffer_2))

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered!")
            break
    
    print("-- Arming")
    await drone.action.arm()
    await asyncio.sleep(5)
    await drone.offboard.set_attitude(Attitude(0.0, 0.0, 0.0, 0))

    print("-- Starting offboard")
    try:
        await drone.offboard.start()
    except OffboardError as error:
        print(f"Starting offboard mode failed with error code: \
              {error._result.result}")
        print("-- Disarming")
        await drone.action.disarm()
        return
    
    await drone.offboard.set_attitude(Attitude(30.0, 0.0, 0.0, 0.2))
    await asyncio.sleep(2.5)

    try:
        await drone.offboard.stop()
    except OffboardError as error:
        print(f"Stopping offboard mode failed with error code: \
              {error._result.result}")
    
    await drone.action.disarm()
    await drone.action.reboot()

async def print_actuator_output_status(drone, data_buffer_1, data_buffer_2):
    async for target in drone.telemetry.actuator_control_target():
        print(target)
        data_buffer_1.append(*target.controls[0:4])
        data_buffer_2.append(*target.controls[4:8])

        if len(data_buffer_1) % 100 == 0:
            data_buffer_1.plot()
        if len(data_buffer_2) % 100 == 0:
            data_buffer_2.plot()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
```

When trying to "disarm" (at the line of `await drone.action.disarm()`),
it gave us an error as follows.

```python3
Waiting for drone to connect...
Drone discovered!
-- Arming
-- Starting offboard
Stopping offboard mode failed with error code:               COMMAND_DENIED
Traceback (most recent call last):
  File "/home/hanna/workspace/github/others/MAVSDK-Python/examples/ex1.py", line 127, in <module>
    loop.run_until_complete(run())
  File "/home/hanna/miniconda3/envs/exp/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/hanna/workspace/github/others/MAVSDK-Python/examples/ex1.py", line 110, in run
    await drone.action.disarm()
  File "/home/hanna/miniconda3/envs/exp/lib/python3.9/site-packages/mavsdk/action.py", line 335, in disarm
    raise ActionError(result, "disarm()")
mavsdk.action.ActionError: COMMAND_DENIED_NOT_LANDED: 'Command Denied Not Landed'; origin: disarm(); params: ()
```

How can we properly exit the offboard mode?

JonasVautherin, 2022-04-07:
I'm a bit confused: why are you trying to disarm while the drone is in the air (see `COMMAND_DENIED_NOT_LANDED`)?

JinraeKim, 2022-04-07:
> I'm a bit confused: why are you trying to disarm while the drone is in the air (see `COMMAND_DENIED_NOT_LANDED`)?

I'm sorry for being confusing.
First, I added the experiment setup in the original post as follows.

> ## Experiment setup
> The vehicle was on a table without propellers for safety. By monitoring telemetry information, we've found that there is no received position data although the vehicle has a GPS module (it is expected as the experiment was conducted indoors).

I'm confused with `COMMAND_DENIED_NOT_LANDED`. The vehicle was on a table, there was no position change, and it did not fly at all (Note: we got rid of propellers for safety). How did it show such an error?

julianoes, 2022-04-07:
@JinraeKim disarm will only disarm if you're landed because you don't want to risk to switch off the motors when flying in-air.

Now the question is why it said "not landed" when you had it on the bench. Any chance you had throttle high?

JinraeKim, 2022-04-08:
> Now the question is why it said "not landed" when you had it on the bench.

Thank you for the clarification.

>  Any chance you had throttle high?
Umm, I think so as we've injected some throttle commands like
> ```python
>     await drone.offboard.set_attitude(Attitude(30.0, 0.0, 0.0, 0.2))
> ```
and the motors were actually rotated as expected.

julianoes, 2022-04-10:
Yes but it's probably enough to trigger in-air.

If throttle doesn't make the props spin, you need to make sure your ESCs are calibrated correctly and the range is set correctly, so that when you arm, and throttle is at idle, they spin!

JinraeKim, 2022-04-11:
> Yes but it's probably enough to trigger in-air.
> 
> If throttle doesn't make the props spin, you need to make sure your ESCs are calibrated correctly and the range is set correctly, so that when you arm, and throttle is at idle, they spin!

Thank you for your answer.

We've resolved this issue by setting the throttle to zero before disarming.

closed 2022-04-11 by JinraeKim