mavlink/MAVSDK-Python #470: `drone.camera.status()` doesn't work | `drone.camera.list_photos(PhotosRange.SINCE_CONNECTION)` always returns an empty list

medric49, 2022-04-17:
I tried to run the `camera.py` example after running

```
make px4_sitl gazebo_typhoon_h480
```

and after manually enabling the camera with
```
camera_trigger start
camera_trigger test
```

and I receive not status update with this function
```
async def print_status(drone):
    async for status in drone.camera.status():
        print(status)
```

I also tested `await drone.camera.list_photos(PhotosRange.SINCE_CONNECTION)` and I always receive an empty list, even when 
`drone.camera.take_photo` seems to work (according to the logs of the simulator shell).

Can someone help me with these issues please?

I also add that `drone.camera.mode()` works fine.. I receive updates about the camera mode.

medric49, 2022-04-17:
It seems that for some methods like `camera.take_photo()` `camera.set_mode()` or `camera.format_storage()`, the communication goes well. But for others (particularly for those that return an answer) like `camera.status()`, `camera.capture_info()` or `camera.list_photos()`, there is a problem.

JonasVautherin, 2022-04-17:
I don't think that `gazebo_typhoon_h480` supports all the camera functionalities. If you try with QGC, probably you will have the same issue, right?

medric49, 2022-04-17:
I see.
No, actually QGC streams perfectly the drone camera.

medric49, 2022-04-17:
Did you succeed to get photos with another drone model?
Which one please?

JonasVautherin, 2022-04-18:
I just don't think that SITL supports that.

medric49, 2022-04-18:
Okay I see.
Finally, since my initial objective was simply to take pictures, regardless of the methods, I switched to this method
```
cam = cv2.VideoCapture("udpsrc port=5600 ! application/x-rtp, payload=96 ! rtph264depay ! h264parse ! avdec_h264"
             " ! decodebin ! videoconvert ! video/x-raw,format=(string)BGR ! videoconvert! appsink emit-signals=true sync=false max-buffers=2 drop=true", cv2.CAP_GSTREAMER)
```

inspired by [this](https://gist.github.com/patrickelectric/443645bb0fd6e71b34c504d20d475d5a)

sibujacob, 2023-05-02:
> Okay I see. Finally, since my initial objective was simply to take pictures, regardless of the methods, I switched to this method
>
> ```
> cam = cv2.VideoCapture("udpsrc port=5600 ! application/x-rtp, [...]

Does this work correctly